carla-simulator/carla #5727: Traffic Sign Bounding Box Labels

dokeefemain, 2022-09-05:
My goal with this software is to train a Multi-Modal object detection model using carla. One thing that I think would be convient about using carla is that I don't need to label my data since I can just run a script and have carla do it automatically. Getting bounding boxes is pretty easy since I'm following this [tutorial](https://carla.readthedocs.io/en/latest/tuto_G_bounding_boxes/) but it says nothing about getting classification labels for the bounding boxes. If you use world.get_level_bbs it returns bounding box objects but I don't get why a label isn't included in the bounding box class. That would litearly make generating a full object detection data set so easy. I know you can do world.get_level_bbs(carla.CityObjectLabel.TrafficSigns) to get the bounding boxes for traffic signs but as far as I can see there is no way to distiguish what the actual traffic sign is. So, if there is a way to get traffic sign bounding boxes and their labels please let me know!

Thank you

amstrudy, 2023-04-13:
Is there an update on this? I am wondering how to do the same thing.

amstrudy, 2023-04-13:
Looking into how to make this modification until I hear back about this--looks like the relevant code is in [`Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Util/BoundingBoxCalculator.cpp`](https://github.com/carla-simulator/carla/blob/be069dda537afa2fff698b70431714489f5c0564/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Util/BoundingBoxCalculator.cpp). 

Are there any implications to adding a new class variable to the FBoundingBox class (found here: [`Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Util/BoundingBox.h'](https://github.com/carla-simulator/carla/blob/fe3cb6863a604f5c0cf8b692fe2b6300b45b5999/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Util/BoundingBox.h))? I'm not very experienced with such large UE4 project code bases.