carla-simulator/carla #5730: Why vehicle spawn with animation "drop from heaven"?

boyjoy1127, 2022-09-06:
Why vehicle spawn with the animation "drop from heaven"?

When a vehicle spawns on a road, it will first appear in the air and then drop on the road.
Could I close this animation? Just spawn it on the road, I prefer.

CARLA version: 0.9.11
Platform/OS: ubuntu 18.04

TR-Monash, 2022-09-12:
You have to reduce the `Z` coordinate value of the spawn point. However, a too low value may cause a collision between the vehicle and the ground and spawning operation may fail.

glopezdiest, 2022-09-22:
AS @TR-Monash, it really isn't an animation, but a way to make sure that no collisions occur between the vehicle and the ground on spawning. Spawn points are a bit above the ground for that exact reason. To avoid it, you can either manually use a waypoint's transform, or reduce the spawn points z value

closed 2022-09-22 by glopezdiest