carla-simulator/carla #5732: Many buildings are not perceived by the LiDAR sensor

vbrebion, 2022-09-06:
Hi!

As part of my research work, I am currently generating perception data with CARLA for learning purposes.
However, it appears that the LiDAR sensor (which is a key component for me) is unable to perceive many buildings, and thus returns erroneous data.

Would it be possible for you to take a look at this issue?

Best,
Vincent

---

**CARLA versions:**
0.9.13; nightly

**Platform/OS:**
Ubuntu 20.04

**Problem you have experienced:**
When I use the LiDAR sensor, many buildings appear to be completely invisible, while they are correctly perceived by other sensors.

- A first example, in the Town01 map:

  [image]
  RGB reference

  [image]
  Depth reference

  [image]
  Projected LiDAR data

  [image]
  Superimposed depth and LiDAR data. Notice the incorrect superimposition for the two large buildings on the right, as well as the two distant buildings on the center-left.

- Another example, in the Town10HD map:

  [image]
  RGB reference

  [image]
  Depth reference

  [image]
  Projected LiDAR data

  [image]
  Superimposed depth and LiDAR data. This time, all the buildings are missing, as well as some other elements, like the flags on the poles.

**Steps to reproduce:**
- Load any map with buildings in them (Town10HD is where the issue seems to appear the most)
- Spawn a LiDAR (I configured it here with a large FOV and many channels for a good visualization of the issue)
- Spawn a depth camera
- Compare the buildings in the projected LiDAR data and in the depth images

**Other information (documentation you consulted, workarounds you tried):**
- I tested the latest development version to see if this issue had already been fixed since the release of the 0.9.13 version, but the issue is still present.
- I know two other similar issues already exist (<https://github.com/carla-simulator/carla/issues/3829> and <https://github.com/carla-simulator/carla/issues/3953>), but they were opened in early 2021, and it appears the source of the error was found at the time (but never merged?)

DarrenWong, 2022-09-21:
I also encounter a similar issue using Town10HD with Carla 0.9.11,
[image]

glopezdiest, 2022-09-23:
Hey @vbrebion, that is indeed the case, thanks for reporting this :+1: . Apparatenly, due to some missconfigurations, some assets are invisible for the LIDAR. We'll be checking all our towns, making sure this is no longer the case, and I'll report back once it's fixed.

heiwang1997, 2022-10-03:
Any update on this? Thanks!

glopezdiest, 2022-10-03:
Yes, we just fixed it last week and it is already in master.

Feel free to reopen the issue if you see any assets that aren't detected by the LIDAR

closed 2022-10-03 by glopezdiest

heiwang1997, 2022-10-03:
@glopezdiest Thank you for your work! This sounds great!

Could you please inform me where can I see the latest changes? For me the latest commit in `master` is still [8854804](https://github.com/carla-simulator/carla/commit/8854804f4d7748e14d937ec763a2912823a7e5f5), which is November last year...

DarrenWong, 2022-11-21:
I tried today's dev branch should be ok to detect the building in Town10. Thanks a lot!

[image]

> I also encounter a similar issue using Town10HD with Carla 0.9.11, [image]