carla-simulator/carla #5741: How can I get the car model information? (revisited)

kino3, 2022-09-09:
Carla 0.9.13, running on Windows

How can I get the car model information (such as vehicle mass, front wheel distance ...) from Carla? Can we get some information by using python API? 

1. I tried [the blue print library](https://carla.readthedocs.io/en/latest/bp_library/), but I can only get surface-level information like color, number_of_wheels. 
1. [This old issue](https://github.com/carla-simulator/carla/issues/1329) explains that we can get some information from UE4Editior. However, this info is too old (2019) to have confidence...
1. As far as I know, there is no public documentation of vehicles specification in Carla.

kino3, 2022-10-11:
### self-resolved

I got some vehicle information by using pythonAPI. Such as,

```
length = vehicle.bounding_box.extent.x * 2
mass = vehicle.get_physics_control().mass
```

#2153 told me that some 3D information unit is not meters but cm so that convert to meters (divide by 100).

closed 2022-10-11 by kino3

fatemehjdi, 2022-12-01:
Hi!
could anyone get the vehicle's information in carla? I want to get the information about tesla model 3 but I do not find them