carla-simulator/carla #5743: Carla exits on startup

Morphlng, 2022-09-09:
## Error Description

I'm trying to run Carla on our GPU server. I've downloaded the pre-built binary version given in the release file, and executed as below:

```bash
./CarlaUE4.sh -RenderOffScreen       // 0.9.12+

// DISPLAY= ./CarlaUE4.sh -opengl for 0.9.11 or lower
```

The output can be seen below:

[image]

As you can see, no error reported, but Carla simply exits. I've followed the Docker-version instruction, same thing happened.

## Platform

Carla version: 0.9.10-0.9.13
OS: Ubuntu 18.04 (display-less server)
GPU: RTX 2080ti
Memory: 256GB

## Compile from source

I've also tried to compile from source. After compiling the UE4 engine, I tried to start it, and it gives message as such:

[image]

This is no big deal, since I don't have any display. So I add -RenderOffScreen, and it crash too:

[image]

From the message, I suspect that's something wrong with the GPU driver, but updating the driver to latest version didn't make any difference. Did anyone meet the same problem? How to solve this other than port to Windows.

YaoWang-Pony, 2023-01-10:
any update about this issue?

YaoWang-Pony, 2023-01-10:
btw where can I find the logs in your screenshots? My `CarlaUE4.sh` exits too without logs, and I don't have any clue to debug it....

Morphlng, 2023-01-10:
> any update about this issue?

I think this issue is happening because your "host system" is actually a docker environment. In my case, I'm running all commands above in a docker ubuntu18.04 environment, I didn't find any solution to bypass directly. Switching to a true host solved this immediately.

> btw where can I find the logs in your screenshots? My `CarlaUE4.sh` exits too without logs, and I don't have any clue to debug it....

These logs are actually produced by Unreal Engine, you'll have to compile the engine from source in order to see those logs.

YaoWang-Pony, 2023-01-11:
Thanks bro. There must be some dependencies missed in my customized image, as I found that Carla works fine with the official one (`carlasim/carla:latest`).

ralwing, 2025-03-03:
In my case mentioned in #6101 there is a missing library in docker:
```
vulkaninfo 
ERROR: [Loader Message] Code 0 : libnvidia-gpucomp.so.550.120: cannot open shared object file: No such file or directory
Cannot create Vulkan instance.
This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.
/root/sdk-build/1.2.131.2/source/Vulkan-Tools/vulkaninfo/vulkaninfo.h:371: failed with ERROR_INCOMPATIBLE_DRIVER
```
Whereas it should be in:
```
/usr/lib/x86_64-linux-gnu/libnvidia-gpucomp.so.*
```

Blyron, 2026-11-12:
As found in the comments, this was a Vulkan library missing inside a custom Docker image; the official image works. For CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) see [Running CARLA in Docker](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_docker.md). Closing.