carla-simulator/carla #5748: Change textures through the API does not works

winterwindwang, 2022-09-12:
CARLA version: Carla 0.9.13 release
Platform/OS: win10
Problem you have experienced: I read an image as a texture, and then use the function `world.apply_color_texture_to_object(target_object, carla.MaterialParameter.Diffuse, texture)` to change the texture of the object. However, after run the function `world.apply_color_texture_to_object(target_object, carla.MaterialParameter.Diffuse, texture)`, the texture of the target object doesn't changed. 
What you expected to happen: The texture of the target object can be updated with the assigned texture.
Steps to reproduce:
I follow the process of [doc](https://carla.readthedocs.io/en/latest/tuto_G_texture_streaming/) to change the texture. Specifically, I first run the CarlaUE4.exe, then I run the `PythonAPI/examples/manual_control.py`. After that, I modify the `PythonAPI/util/apply_texture.py` with the following code 

`
    target_object = "BP_ChargerCop2020_C_2147479460"
    print('Altering texture for object: ' + target_object)
 
    # tex_path is the a jpg image
    image = Image.open(tex_path)
    image = ImageOps.fit(image, (2048,2048))

    height = image.size[1]
    width = image.size[0]
    texture = carla.TextureColor(width, height)
    for x in range(0, width):
        for y in range(0, height):
            color = image.getpixel((y, x))
            r = int(color[0])
            g = int(color[1])
            b = int(color[2])
            a = 255
            texture.set(x, height - y - 1, carla.Color(r,g,b,a))
            # texture.set(x, y, carla.Color(r, g, b, a))

    world.apply_color_texture_to_object(target_object, carla.MaterialParameter.Diffuse, texture)
`
I get the targe object name by  `print(world.get_names_of_all_objects())`,
`
 'BP_Lamppost_tall31_38', 'BP_Lamppost_tall32_41', 'BP_Lamppost_tall33_44', 'BP_Lamppost_tall34_47', 'BP_Lamppost_tall35_53', 'BP_Lamppost_tall36_56', 'BP_Lamppost_tall37_59', 'BP_Lamppost_tall38_62', 'BP_Lamppost_tall39_65', 'BP_Lamppost_tall3_8', 'BP_Lamppost_tall40_2', 'BP_Lamppost_tall41_2', 'BP_Lamppost_tall42', 'BP_Lamppost_tall43', 'BP_Lamppost_tall44', 'BP_Lamppost_tall45', 'BP_Lamppost_tall46', 'BP_Lamppost_tall47_10', 'BP_Lamppost_tall48_13', 'BP_Lamppost_tall49_16', 'BP_Lamppost_tall4_2', 'BP_Lamppost_tall50_19', 'BP_Lamppost_tall51', 'BP_Lamppost_tall52', 'BP_Lamppost_tall53', 'BP_Lamppost_tall54', 'BP_Lamppost_tall55', 'BP_Lamppost_tall56', 'BP_Lamppost_tall57_2', 'BP_Lamppost_tall58', 'BP_Lamppost_tall59', 'BP_Lamppost_tall5_5', 'BP_Lamppost_tall60', 'BP_Lamppost_tall61', 'BP_Lamppost_tall62', 'BP_Lamppost_tall63', 'BP_Lamppost_tall64', 'BP_Lamppost_tall65', 'BP_Lamppost_tall66', 'BP_Lamppost_tall67', 'BP_Lamppost_tall68', 'BP_Lamppost_tall69', 'BP_Lamppost_tall6_8', 'BP_Lamppost_tall70', 'BP_Lamppost_tall71', 'BP_Lamppost_tall72', 'BP_Lamppost_tall73', 'BP_Lamppost_tall74', 'BP_Lamppost_tall75', 'BP_Lamppost_tall76', 'BP_Lamppost_tall77', 'BP_Lamppost_tall78', 'BP_Lamppost_tall79', 'BP_Lamppost_tall7_11', 'BP_Lamppost_tall80', 'BP_Lamppost_tall81', 'BP_Lamppost_tall82', 'BP_Lamppost_tall83', 'BP_Lamppost_tall84', 'BP_Lamppost_tall85', 'BP_Lamppost_tall86', 'BP_Lamppost_tall87', 'BP_Lamppost_tall88', 'BP_Lamppost_tall89', 'BP_Lamppost_tall8_2', 'BP_Lamppost_tall9', 'BP_Lamppost_tall90', 'BP_Lamppost_tall91', 'BP_Lamppost_tall92', 'BP_Lamppost_tall93', 'BP_Lamppost_tall94', 'BP_Lamppost_tall95', 'BP_Lamppost_tall96', 'BP_Lamppost_tall_2', 'BP_Lamppost_short10_2', 'BP_Lamppost_short11_18', 'BP_Lamppost_short12', 'BP_Lamppost_short13_12', 'BP_Lamppost_short14_15', 'BP_Lamppost_short15_2', 'BP_Lamppost_short16_5', 'BP_Lamppost_short17_8', 'BP_Lamppost_short18_11', 'BP_Lamppost_short19_14', 'BP_Lamppost_short2', 'BP_Lamppost_short20_17', 'BP_Lamppost_short21_20', 'BP_Lamppost_short22_23', 'BP_Lamppost_short23_26', 'BP_Lamppost_short24_29', 'BP_Lamppost_short25_32', 
'BP_Lamppost_short26_35', 'BP_Lamppost_short27_38', 'BP_Lamppost_short28_41', 'BP_Lamppost_short29_44', 'BP_Lamppost_short3', 'BP_Lamppost_short30_47', 'BP_Lamppost_short31_50', 'BP_Lamppost_short32_53', 'BP_Lamppost_short33_56', 'BP_Lamppost_short34_59', 'BP_Lamppost_short35_2', 'BP_Lamppost_short36_5', 'BP_Lamppost_short37_8', 'BP_Lamppost_short38_11', 'BP_Lamppost_short39_14', 'BP_Lamppost_short4', 'BP_Lamppost_short40_17', 'BP_Lamppost_short5', 'BP_Lamppost_short6', 'BP_Lamppost_short7', 'BP_Lamppost_short8_11', 'BP_Lamppost_short9', 'BP_Lamppost_short_2', 'BP_HighwayLight_multiple2', 'BP_HighwayLight_multiple3_2', 'BP_HighwayLight_multiple4', 'BP_HighwayLight_multiple5_2', 'BP_HighwayLight_multiple_2', 'BP_Hotel_Lights10_16', 'BP_Hotel_Lights12_18', 'BP_Hotel_Lights2_1', 'BP_Hotel_Lights3_3', 'BP_Hotel_Lights4_5', 'BP_Hotel_Lights5', 'BP_Hotel_Lights6_8', 'BP_Hotel_Lights7_10', 'BP_Hotel_Lights8_12', 'BP_Hotel_Lights9_14', 'BP_Hotel_Lights_5', 'BP_Hotel_Lights13_2', 'BP_Hotel_Lights14', 'BP_Hotel_Lights15', 'BP_Hotel_Lights16', 'BP_Hotel_Lights17', 'BP_Hotel_Lights18', 'BP_Hotel_Lights19', 'BP_Hotel_Lights20', 'BP_Hotel_Lights21', 'BP_Hotel_Lights22', 'CollisionSensor_2147479410', 'GnssSensor_2147479408', 'InertialMeasurementUnit_2147479407', 'LaneInvasionSensor_2147479409', 'BP_Stop2_2', 'BP_Stop3_8', 'BP_Stop4_11', 'BP_Yield_2', 'BP_Skyscraper_02_v04_Opt2_5', 'BP_Skyscraper_02_v04_Opt3', 'BP_Skyscraper_02_v04_Opt_2', 'BP_Skyscraper_02_v01_Opt_Child2', 'BP_Skyscraper_02_v01_Opt_Child_2', 'BP_Skyscraper_02_v03_Opt_2', 'BP_Skyscraper_02_v02_Opt_2', 'BP_CarlaCola2_5', 'BP_CarlaCola3_8', 'BP_CarlaCola_2', 'BP_TrashCan03_5', 'BP_TrafficLightNew_T10_master_mediumBIG_rsc2', 'BP_TrafficLightNew_T10_master_mediumBIG_rsc_2', 'BP_TrafficLightNew_T10_master_largeBIG_rsc11', 'BP_TrafficLightNew_T10_master_largeBIG_rsc12', 'BP_TrafficLightNew_T10_master_largeBIG_rsc13', 'BP_TrafficLightNew_T10_master_largeBIG_rsc14', 'BP_TrafficLightNew_T10_master_largeBIG_rsc2', 'BP_TrafficLightNew_T10_master_largeBIG_rsc3', 'BP_TrafficLightNew_T10_master_largeBIG_rsc4', 'BP_TrafficLightNew_T10_master_largeBIG_rsc5', 'BP_TrafficLightNew_T10_master_largeBIG_rsc6', 'BP_TrafficLightNew_T10_master_largeBIG_rsc7', 'BP_TrafficLightNew_T10_master_largeBIG_rsc8', 'BP_TrafficLightNew_T10_master_largeBIG_rsc9', 'BP_TrafficLightNew_T10_master_largeBIG_rsc_2', 'SceneCaptureCamera_2147479366', 'BP_ChargerCop2020_C_2147479460']
`

After running `PythonAPI/util/apply_texture.py,` the texture `BP_ChargerCop2020_C_2147479460` of  does not change. 

Additionally, I tried to modify the other object's texture but also failed.

Therefore, I want to know how to correctly change the texture of the target object.

neilsambhu, 2023-04-19:
@winterwindwang I am also unable to modify the texture of a vehicle using the API. I have been able to modify a TGA file from an export from Unreal Engine compiled from source for CARLA. I used GIMP to modify the TGA file with X's. I applied the texture following https://carla.readthedocs.io/en/latest/tuto_G_texture_streaming/ and changed the Diffuse Textures of the Parameter Groups of MI_BodyWork_Ambulance. Regardless of the X's being red, white, or black in GIMP, I always see white X's.

winterwindwang, 2023-05-31:
> I am also stuck at the same problem with stop sign.

I can change some of assets in the scene, e.g., SM_bannerSingle14 and so on. However, when I load some custom import 3D object (e.g., car), I can't change its texture, although the code is run without error.

winterwindwang, 2023-05-31:
> @winterwindwang I am also unable to modify the texture of a vehicle using the API. I have been able to modify a TGA file from an export from Unreal Engine compiled from source for CARLA. I used GIMP [...]

Yes, I can modify some scene object with the API. However, modifying the custom import object seem doesn't work.

rg1999, 2023-08-30:
Hi, 
Any update on this, still It's not working for assets created in roadrunner. or is there any other way to change color/texture of an asset created in roadrunner then imported in carla simulator.

FriedrichZimmer, 2024-04-10:
I also had this problem, that I was able to update only certain textures by API during runtime.
After some experimenting with the Editor, I found out, that only only Materials or Material Instances with a Global Texture Parameter called "BaseColor" are updateable by API.
I fixed this by creating a new material instance based on the material "M_Material_Master" and applied my textures there.

For the Stop sign, the Problem is (at least in Carla 9.15) that the material instance "MI_Stops" is having the material "M_MasterSigns" as parent which is using other labels for some reason. I guess here renaming the Parameters "SpeedSign_d" into "BaseColor" and SpeedSign_n" into "Normal" is the best solution.

[image]

caixxuan, 2024-04-16:
I was wondering how to know the target_object.name of the certain vehicle if i want to apply to a vehicle in the map?

neilsambhu, 2024-04-17:
You have to build CARLA from source using Unreal Engine. Unreal Engine will
show you the assets to make up the car.

Best regards,
Neil Sambhu