carla-simulator/carla #5749: CARLA takes 100% GPU without running a simulation

RaviBeagle, 2022-09-12:
Hello CARLA-Developers,

Please see this issue below

_CARLA version_: 0.9.12
_Platform/OS_: Ubuntu 20.04 on HP ZBook Z8, NVIDIA RTX A3000 GPU, NVIDIA-SMI 510.85.02    Driver Version: 510.85.02 
__Problem you have experienced:_ CARLA server takes 100% GPU without even running a simulation
What you expected to happen_: It should not consume GPU resource until we run a simulation using a client
_Steps to reproducce_: Just run ./CarlaUE4.sh -carla-server -RenderOffScreen on such a system and I see the GPU at constant 99% utilization and laptop fan runs continuously.

RaviBeagle, 2022-09-29:
Is this normal ?

RaviBeagle, 2022-10-03:
After some analyzing myself, I found that CARLA by default loads the Town10HD Map and runs asynchronously which is causes 100% GPU utilization.

This level of utilization is also true when running CARLA with any other town map at a high fps (say 1000 or even 100 fps) in synchronous mode. 
When I run the simulation at 10fps the GPU utilization drops between 20-40%

closed 2022-10-03 by RaviBeagle

huangmingguang, 2024-11-06:
hi, I have the same problem. How to do it to solve this problem.