carla-simulator/carla #5761: Not running on ubuntu 22.04

resistorfail, 2022-09-16:
I am trying to install Carla 0.9.13 on Ubuntu 22.04. I am following the Debian Carla installation instructions as mentioned [here.](https://carla.readthedocs.io/en/latest/start_quickstart/#a-debian-carla-installation)
First I run:
`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9` 
It warns me that apt-key is deprecated so I use gpg to make sure it works. Moving on, I enter:
` sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"`
It makes it past the prompt (I press enter) and runs through hit 1-10 and gets a 404 on hit 8. It then spits out the following error(s):
```
E: The repository 'http://dist.carla.org/carla jammy Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
I'm used to seeing errors, most of which are benign. I hoped that this wouldn't be a problem and moved on.
I ran `sudo apt-get update` which responded as usual except with an error about carla
I understand I should probably accept defeat at this point, but I cross my fingers and run:
`sudo apt-get install carla-simulator`
It couldn't find that
I try `sudo apt install carla` which worked
I try to open Carla, but it does nothing.

Note: the reason a lot of the error codes (specifically the ones for `apt-get update` and `apt get install carla-simulator` and even upon trying to recreate `apt install carla` is when i tried to recreate these errors, the only thing that any of them responded with was
```
E: Malformed entry 3 in list file /etc/apt/sources.list.d/archive_uri-http_dist_carla_org_carla-jammy.list (Component)
E: The list of sources could not be read.

```

RaviBeagle, 2022-09-20:
Did you see this in the documentation: "The Debain package is available for both Ubuntu 18.04 and Ubuntu 20.04, however the officially supported platform is Ubuntu 18.04." ?

resistorfail, 2022-09-22:
yes. I was, however wondering if there was a way to get it to work otherwise.

RaviBeagle, 2022-09-23:
You could try running the Docker version:  https://carla.readthedocs.io/en/latest/build_docker/

xqgex, 2022-10-01:
On Ubuntu 22.04, the output of the command `$(lsb_release -sc)` is `jammy`
If you will go to http://dist.carla.org/carla/dists/ you will see that there are only 'focal' (Ubuntu 20.04) and 'bionic' (Ubuntu 18.04)
So you can remove the repository that you have added as it will give you nothing but a 404 error.

The command `apt install carla` installed on your system something else https://packages.ubuntu.com/jammy/carla
https://kx.studio/Applications:Carla

> Carla is a fully-featured modular audio plugin host

xqgex, 2022-10-01:
A few notes as it might help someone (who understands that this is only an experiment because the software was designed for Ubuntu 18.04).

# In order to install Carla simulator on Ubuntu 22.04

A partially working instructions

## Install the preconditions

1. `sudo apt install libomp5`
   This will resolve the error:
   ```
   /opt/carla-simulator/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping: error while loading shared libraries: libomp.so.5: cannot open shared object file: No such file or directory
   ```
1. `sudo apt install python3-open3d`
   This will resolve the error:
   ```
   ERROR: Could not find a version that satisfies the requirement open3d (from versions: none)
   ERROR: No matching distribution found for open3d
   ```
1. `pip3 install --upgrade pip`
1. `pip3 install --user pygame numpy future matplotlib open3d Pillow`

If you are using Ubuntu 22.04 with Python 3.10+, don't run:
`pip3 install -r /opt/carla-simulator/PythonAPI/examples/requirements.txt`
because you will get the following error:
```
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for numpy
Failed to build numpy
ERROR: Could not build wheels for numpy, which is required to install pyproject.toml-based projects
```

## Install Carla

1. `sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9`
1. `sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla focal main"`
1. `sudo apt update`
1. `sudo apt install carla-simulator`

## Run the simulation server

`/opt/carla-simulator/CarlaUE4.sh -opengl4 -carla-server -benchmark -fps=10`

:tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada:

## Run the simulation client

This is where I stopped, as I am getting the following error:
```
python3.10 crashed with SIGSEGV in "_Py_NoneStruct()"
```

```
$ python3
Python 3.10.6 (main, Aug 10 2022, 11:40:04) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import carla
Segmentation fault (core dumped)
$
```

RaviBeagle, 2022-10-14:
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
[sudo] password for sxv1kor: 
Executing: /tmp/apt-key-gpghome.ZATpndfcYt/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
gpg: keyserver receive failed: No name

Does not work on Ubuntu 20.04 for me

xqgex, 2023-02-24:
> sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9 [sudo] password for sxv1kor: Executing: /tmp/apt-key-gpghome.ZATpndfcYt/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9 gpg: keyserver receive failed: No name
> 
> Does not work on Ubuntu 20.04 for me

Maybe a local connectivity error?
Try to access the server with `curl`: `curl http://keyserver.ubuntu.com | grep description`

> ```
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
>100  9817  100  9817    0     0  29445      0 --:--:-- --:--:-- --:--:-- 29480
>      <meta name="description" content="OpenPGP Keyserver">
> ```

Also try to use the full url:
```
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 1AF1527DE64CB8D9
```

Expected output:
```
[sudo] password for <user_name>: 
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.BiEpONolIg/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 1AF1527DE64CB8D9
gpg: key 1AF1527DE64CB8D9: public key "Carla Simulator Team <carla.simulator@gmail.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
``

bisalgt, 2023-11-14:
Hello everyone. Anyone able to get carla working with ubuntu222.04. Thank you in advance.

soumya997, 2024-01-04:
Hi @bisalgt, looking at the above discussion using docker seems like the only feasible solution.

bisalgt, 2024-01-05:
Yeah. Maybe I think. I didnot consider docker for myself because i thought I might be having issues in future with the carla in docker along with display. So, I downgraded my system to Ubuntu 20.04. Its working fine for now.

Blyron, 2024-04-26:
We have updated Build docs to use CARLA with Ubuntu 22.
Are you okay with closing this issue?

bisalgt, 2024-04-27:
Thats a great news @Blyron . Thank you for letting us know in the issue. I think its fine to close the issue. Thanks again to the development team.

closed 2024-05-03 by Blyron