carla-simulator/carla #5765: Unreal engine not detected

arsalan0302, 2022-09-17:


arsalan0302, 2022-09-17:
I am facing the following error when I run 'make launch' command, I followed  windows build carla Docs step by step but still same error happens.
please share your ideas.
Thank You!
[image]

Blyron, 2024-07-12:
You have to set Enviorment variable of your Unreal Engine Path
You haven't set properly the ENV Variable
Set Unreal Engine environment variable
It is necessary to set an environment variable so that CARLA can find the Unreal Engine installation folder. This allows users to choose which specific version of Unreal Engine is to be used. If no environment variable is specified, then CARLA will search for Unreal Engine in the windows registry and use the first version it finds there.

To set the environment variable:

Open Windows Control Panel and go to Advanced System Settings or search for Advanced System Settings in the Windows search bar.
On the Advanced panel open Environment Variables....
Click New... to create the variable.
Name the variable UE4_ROOT and choose the path to the installation folder of the desired Unreal Engine installation.

closed 2024-07-12 by Blyron