carla-simulator/carla #5767: Custom map loading in carla-simulator fails saying corrupted data, serialization failed (missing .uexp and .uasset)

ranimg, 2022-09-18:


ranimg, 2022-10-10:
Hi All, any inputs on this please ?

Dragonking3x, 2022-10-14:
What I found out was, that you can't copy the files from the Carla branch to the Release one.
Carla branch = clone this git
Release one = 
[image]

You can make your own package with your map with "make package".
Don't forget to add your map to the build settings.
Unreal Editor -> Edit -> Project settings
Project -> Packaging -> Packaging -> (Schow Advanced) -> List of maps to include in package to build

If you make the package, there is a point at which you get no command line output for 30-60 minutes.

Blyron, 2026-11-12:
A custom map has to be cooked into the package. The supported way is a content pack, see [Content packs](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_packs.md) and [Custom maps](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_maps.md). Closing.

closed 2026-11-12 by Blyron