carla-simulator/carla #5769: Fatal Error on 2nd call to load_world

dk42, 2022-09-19:
CARLA version: 0.9.13 and 0.9.12
Platform/OS: Windows
Problem you have experienced: Loading maps leads to "Fatal Error!"
What you expected to happen: Maps get loaded
Steps to reproduce:
1. Start Carla with quality-level=Low
2. Run the following Script:
```python
from carla import Client

client = Client('localhost', 2000)
client.set_timeout(60)

print('Loading TOWN04...')
client.load_world('TOWN04')

print('Loading DONE')

print('Loading TOWN01...')
client.load_world('TOWN01')
```

The error message is the following:
```
Fatal error!

Unhandled Exception: EXCEPTION_ACCESS_VIOLATION 0x0000000000000000

0x0000000000000000 UnknownFunction []
0x00007ff6694ffc1c CarlaUE4-Win64-Shipping.exe!UTexture2D::GetNumResidentMips() [C:\UE_4.26\Engine\Source\Runtime\Engine\Private\Texture2D.cpp:213]
0x00007ff6688ddb26 CarlaUE4-Win64-Shipping.exe!FLandscapeRenderSystem::FGetSectionLODBiasesTask::AnyThreadTask() [C:\UE_4.26\Engine\Source\Runtime\Landscape\Public\LandscapeRender.h:640]
0x00007ff6688e86ac CarlaUE4-Win64-Shipping.exe!TGraphTask<FLandscapeRenderSystem::FGetSectionLODBiasesTask>::ExecuteTask() [C:\UE_4.26\Engine\Source\Runtime\Core\Public\Async\TaskGraphInterfaces.h:889]
0x00007ff66763280a CarlaUE4-Win64-Shipping.exe!FTaskThreadAnyThread::ProcessTasks() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Async\TaskGraph.cpp:1066]
0x00007ff6676334a8 CarlaUE4-Win64-Shipping.exe!FTaskThreadAnyThread::ProcessTasksUntilQuit() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Async\TaskGraph.cpp:888]
0x00007ff66763562c CarlaUE4-Win64-Shipping.exe!FTaskThreadBase::Run() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Async\TaskGraph.cpp:541]
0x00007ff6677d267b CarlaUE4-Win64-Shipping.exe!FRunnableThreadWin::Run() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Windows\WindowsRunnableThread.cpp:86]
0x00007ff6677cdcc1 CarlaUE4-Win64-Shipping.exe!FRunnableThreadWin::GuardedRun() [C:\UE_4.26\Engine\Source\Runtime\Core\Private\Windows\WindowsRunnableThread.cpp:35]
0x00007ffc42787034 KERNEL32.DLL!UnknownFunction []
0x00007ffc43ca26a1 ntdll.dll!UnknownFunction []

Crash in runnable thread TaskGraphThreadNP 0
```

Any idea why this happens? Or why this doesn't seem to happen with quality-level=Epic? How can I debug the problem further? 

Thanks!

Victorrsl, 2025-10-06:
LOD（Level of Detail） Bug in Low Quality Mode: When using -quality-level=Low, UE4 significantly reduces the loading of texture mipmaps. The terrain and textures in map(Town04) are complex. Certain texture objects fail to initialize correctly in low quality mode, causing GetNumResidentMips() to access null objects during its call.
This may require you to manually modify the config file to lower the rendering quality, rather than simply using this command to achieve the desired result. This is clearly a bug in UE4, and it may be fixed in future versions.

Blyron, 2026-11-12:
This report is about an old CARLA release on Unreal Engine 4, which is no longer maintained. Closing as obsolete. Please try CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`). If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron