carla-simulator/carla #5770: Could NOT find SDL2

jiankebe, 2022-09-19:
How to set up SDL2 ？

Environment: 
carla source code: master 
vs: 2019
windows: 11
cmake: 3.24.2
make: 3.81

but i have a problem like: 
'D:/carla/carla/Build/recast-0b13b0-src' is on a file system that doesnot record ownership
To add an exception for this directory, call:

        git config --global --add safe.directory D:/carla/carla/Build/recast-0b13b0-src
    -[install_recast]: Creating "D:\carla\carla\Build\recast-0b13b0-src\build\"
    -[install_recast]: Generating build...
-- The C compiler identification is MSVC 19.29.30146.0
-- The CXX compiler identification is MSVC 19.29.30146.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: opengl32
CMake Error at C:/Program Files/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find SDL2 (missing: SDL2_LIBRARY SDL2_INCLUDE_DIR
  SDL2_SDLMAIN_LIBRARY)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  RecastDemo/cmake/FindSDL2.cmake:152 (find_package_handle_standard_args)
  RecastDemo/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "D:/carla/carla/Build/recast-0b13b0-src/build/CMakeFiles/CMakeOutput.log".

    -[install_recast]: [CMAKE ERROR] An error ocurred while executing the cmake.
    -[install_recast]: [CMAKE ERROR] Possible causes:
    -[install_recast]:                - Make sure "CMake" is installed.
    -[install_recast]:                - Make sure it is available on your Windows "path".
    -[install_recast]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 错误 1

Kurt-Morrison, 2022-10-23:
Have you solved this problem about carla missing SDL2?  I met the same issue。thx。

jiankebe, 2023-03-10:
> Have you solved this problem about carla missing SDL2? I met the same issue。thx。

Unfortunately, not yet! 

I change os to windows 10 to resolve it

ppfei666, 2024-01-25:
Have you solved this problem ? I have the same problem.

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron