carla-simulator/carla #5778: wheel run freely but vehicle can not move

YanggangZhang, 2022-09-22:
CARLA version:0.9.12
Platform/OS:18.04

I  made a simulation scene by roadrunner and import it into carla with UE roadrunner plugin freely. when i run "python generate_traffic.py" to test this map, some errors happened. the vehicle wheel overlap with road surface but run freely and the vehicle can not move. someone has meet with the same proplem too? how to fix it? thx! 

problem:
[image]

road collision setting:
[image]

boxwayne, 2023-04-10:
Had you solved this problem? I got the same problem and look for a solution.

yu-ju-chiu, 2023-05-18:
Have you solved the issue?
The CarlaUE4 gets stuck every time, when I run python file.

gsmario, 2024-05-17:
Hello, sorry for the long time answer, do you still need help with this?
Let me know, I'm trying to reproduce this problem but i can't.

gsmario, 2024-06-07:
I'm closing this issue, feel free to reopen it if you need help.
Thanks.

closed 2024-06-07 by gsmario