carla-simulator/carla #5779: Pedestrian bounding boxes are lagging behind the pedestrian

makaveli10, 2022-09-23:
CARLA version: 0.9.12
Platform/OS: Ubuntu 20.04
Hello Carla Community,
I have been trying to create some synthetic datasets out of carla and so far its been great.
The bounding boxes for pedestrians which are more to the side of the camera rather than front are lagging behind the pedestrian. For e.g.
[image]
Has anybody else experienced this issue.

P.S. I am using sync mode.

amsvenkat, 2023-01-03:
Yes, i am facing the same issue and stuck with it in 0.9.11.

Sondosmohamed1, 2023-09-13:
Hi ,did you solve this problem? i have the same issue 
[image]