carla-simulator/carla #5786: make: *** [CarlaUE4Editor] 错误 6 （How to solve this problem ）

jiaqi-lian, 2022-09-26:
Building 8 actions with 32 processes...
  [1/8] Module.Carla.gen.5_of_7.cpp
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(61): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(161): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/TaggedComponent.h(83): error C2504: “FHierarchicalStaticMeshSceneProxy”: 未定义基类
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/TaggedComponent.h(87): error C3668: “FTaggedHierarchicalStaticMeshSceneProxy::GetViewRelevance”: 包含重写说明符“override”的方法没有重写任何基类方法
  [2/8] Module.Carla.3_of_3.cpp
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(61): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(161): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/CarlaHUD.h(53): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/CarlaHUD.h(54): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(43): warning C4996: 'UWheeledVehicleMovementComponent::bReverseAsBrake': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(33): warning C4996: 'AWheeledVehicle::AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(43): warning C4996: 'AWheeledVehicle::GetVehicleMovementComponent': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(49): warning C4996: 'AWheeledVehicle::~AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(51): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(139): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(139): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(147): warning C4996: 'UWheeledVehicleMovementComponent::Wheels': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(164): warning C4996: 'UWheeledVehicleMovementComponent::WheelSetups': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(168): warning C4996: 'UVehicleWheel': PhysX is deprecated. Use the UChaosVehicleWheel from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(172): warning C4996: 'UWheeledVehicleMovementComponent::WheelSetups': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(140): warning C4996: 'AWheeledVehicle::GetVehicleMovementComponent': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(229): warning C4996: 'UWheeledVehicleMovementComponent::Wheels': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(233): warning C4996: 'UVehicleWheel::SteerAngle': PhysX is deprecated. Use the UChaosVehicleWheel from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(229): warning C4996: 'AWheeledVehicle::GetVehicleMovementComponent': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp(285): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
[... cut ...]
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(476): warning C4458: “Role”的声明隐藏了类成员
  D:\Program Files\Epic Games\UE_4.26\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(541): note: 参见“AActor::Role”的声明
  C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(491): warning C4458: “Role”的声明隐藏了类成员
  D:\Program Files\Epic Games\UE_4.26\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(541): note: 参见“AActor::Role”的声明
make: *** [CarlaUE4Editor] 错误 6`

1604148641, 2022-09-26:
I just got the same error while building Carla which commit hash is 8854804f4d7748e14d937ec763a2912823a7e5f5.
Have you ever try the previous versions?

1604148641, 2022-09-27:
l solve this issue afier using the specific fork of UE4 for carla. My carla version is 0.9.13.

> **Starting with version 0.9.12, CARLA uses a modified fork of Unreal Engine 4.26. This fork contains patches specific to CARLA**.
> 
> In a terminal, navigate to the location you want to save Unreal Engine and clone the carla branch:
> 
>     git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git .

U can get this modified fork of UE4 after binding your github account with your epic account .

Alphabaijinde, 2022-10-21:
> l solve this issue afier using the specific fork of UE4 for carla. My carla version is 0.9.13.
>
>
> U can get this modified fork of UE4 after binding your github account with your epic account .

Creating makefile for CrashReportClient (no existing makefile)
ERROR: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
make: *** [Makefile:327: CrashReportClient-Linux-Shipping] Error 6

I met this issue

MoolixLin, 2022-11-06:
Have you solved this problem？ Which way you choose? I meet the same problem as you.

Alphabaijinde, 2022-11-06:
这是来自QQ邮箱的假期自动回复邮件。您好，我已经收到您的邮件，会尽快回复的。

demirciogl, 2023-05-16:
have you solved this problem?

closed 2024-05-17 by MarcelPiNacy-CVC