carla-simulator/carla #5787: Assertion failed when importing a map

vttitta, 2022-09-26:


DanyThodms, 2022-12-15:
I have encountered the same problem and expect to get a solution.

FBanani, 2023-06-29:
> CARLA version: 0.9.13 built from source version Platform/OS: Windows 10 Problem you have experienced: I'm trying to import a map to CARLA using this tutorial https://carla.readthedocs.io/en/0.9.13/tuto_M_add_map_source/ => putting my FBX and XODR files to import-folder and make import. Import fails with Assertion failed: _length > 0.0 What you expected to happen: Import functional map to Carla Other information (documentation you consulted, workarounds you tried): I originally thought it was because my map is pretty large (about 5kmx5km) and reduced it to about 1kmx1km. Still same error. Still too big?
> 
> Edit: Tried to import just osm (with this https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/), and receive the same error. The problem then seems to not be with the FBX-file

I have this issue. Have you had a chance to solve it?