carla-simulator/carla #5790: The locations of crosswalk`s polygon which get from PythonAPI seems not correct.

1604148641, 2022-09-27:
The locations of crosswalk`s polygon which get from PythonAPI seems not correct. How can I fix these coordinates cause I want to generate some dataset for crosswalk throught this API.

Here is the code I draw the outline of crosswalk:
```python
# get coordinates 
crosswalks_list = world.get_map().get_crosswalks()

# split up
list_of_groups = zip(*(iter(crosswalks_list),) * 5)
cw_polygons = [list(i) for i in list_of_groups]
count = len(crosswalks_list) % 2
cw_polygons.append(crosswalks_list[-count:]) if count != 0 else cw_polygons

# draw line
for cw in cw_polygons:
    cw[0].z = 0.5
    cw[1].z = 0.5
    cw[2].z = 0.5
    cw[3].z = 0.5
    client.get_world().debug.draw_line(cw[0], cw[1], thickness=0.1, color=carla.Color(255, 0, 0), life_time=0)
    client.get_world().debug.draw_line(cw[1], cw[2], thickness=0.1, color=carla.Color(255, 0, 0), life_time=0)
    client.get_world().debug.draw_line(cw[2], cw[3], thickness=0.1, color=carla.Color(255, 0, 0), life_time=0)
    client.get_world().debug.draw_line(cw[3], cw[0], thickness=0.1, color=carla.Color(255, 0, 0), life_time=0)
```

The final result:
[image]
[image]

ccysway, 2023-08-23:
Hi,
    I met the same problem.It seems good in https://odrviewer.io/,but not right in carla-0.9.14.
[image]
[image]

rodrigoqueiroz, 2024-05-14:
I'm having the same issue. All crosswalks are off in some degree.

kevin1zc, 2025-05-13:
Issue still exists in Carla 0.9.15.2

GoodarzMehr, 2025-08-07:
This may help for Town03-Town10HD:

```Python
objects = world.get_environment_objects()

crosswalks = [obj for obj in objects if '_Crosswalk_' in obj.name]

bbox = crosswalk.bounding_box.get_local_vertices()

crosswalk_box = [bbox[0], bbox[2], bbox[6], bbox[4], bbox[0]]
```

Aperiss, 2025-08-21:
This issue should be resolved in, https://github.com/carla-simulator/carla/pull/9173. Let me know if this works for you