carla-simulator/carla #5793: Export maps from Roadrunner to UE4

Into-sight, 2022-09-28:
HI Team,

I have successfully imported a simple map into UE4, but when I tried to import a complex map (1000mX1000m, with lots of trees and buildings), I failed.

CARLA version:0.9.13 (latest)
Unreal Engine : 4.26
Platform/OS:Windows 10
What you expected to happen: Map ingestion into UE4

Error:
C:\carla>make import ARGS="--package MapPackage --no-carla-materials"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --package MapPackage --no-carla-materials
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\carla\Build\" --zlib-install-dir "C:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\carla\Build\" --toolset msvc-14.2 --version 1.72.0 -j 16
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
已复制         1 个文件。
已复制         1 个文件。
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
已复制         1 个文件。
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "C:\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
已复制         1 个文件。
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "C:\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
已复制         1 个文件。
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########
[... 116 lines cut ...]
Partition of walkable surfaces...
Remove walkable surfaces by agent raius...
Defining areas by material
Trace and simplify region contours...
Build polygons mesh from contours...
Create detail mesh...
Create Detour data from Recast poly mesh...
Creating navigation data...
Saving...
Copying "C:\carla\Util\DockerUtils\dist\MyCity2.bin" to "C:\carla\Unreal\CarlaUE4\Content\package\Maps\MyCity2\Nav\MyCity2.bin"
Traceback (most recent call last):
  File "C:\carla\Util\BuildTools\Import.py", line 637, in <module>
    main()
  File "C:\carla\Util\BuildTools\Import.py", line 634, in main
    import_assets_from_json_list(json_list, args.batch_size)
  File "C:\carla\Util\BuildTools\Import.py", line 454, in import_assets_from_json_list
    import_assets(package_name, dirname, props, maps, 0, 0, 0)
  File "C:\carla\Util\BuildTools\Import.py", line 390, in import_assets
    invoke_commandlet(commandlet_name, commandlet_arguments)
  File "C:\carla\Util\BuildTools\Import.py", line 211, in invoke_commandlet
    subprocess.check_call(command, shell=True)
  File "C:\Users\77550\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['D:\\UnrealEngine/Engine/Binaries/Win64/UE4Editor', 'C:\\carla\\Unreal\\CarlaUE4\\CarlaUE4.uproject', '-run=ImportAssets', '-importSettings="C:\\carla\\importsetting.json"', '-nosourcecontrol', '-replaceexisting']' returned non-zero exit status 3221225477.
make: *** [import] 错误 1

ronyshaji, 2022-12-15:
> HI Team,
> 
> I have successfully imported a simple map into UE4, but when I tried to import a complex map (1000mX1000m, with lots of trees and buildings), I failed.
> 
> CARLA version:0.9.13 (latest) Unreal Engine : 4.26 Platform/OS:Windows 10 What you expected to happen: Map ingestion into UE4
> 
> Error: C:\carla>make import ARGS="--package MapPackage --no-carla-materials" -[Setup]: [Batch params]: --boost-toolset msvc-14.2 --package MapPackage --no-carla-materials -[Setup]: Asynchronous jobs: 16 -[Setup]: Boost toolset: msvc-14.2 -[Setup]: Install directory: "C:\carla\Build" -[Setup]: Installing zlib... -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build" -[install_zlib]: A zlib installation already exists. -[install_zlib]: Delete "C:\carla\Build\zlib-install" if you want to force a rebuild. -[install_zlib]: Exiting... -[Setup]: Installing libpng... -[install_libpng]: [Batch params]: --build-dir "C:\carla\Build" --zlib-install-dir "C:\carla\Build\zlib-install" -[install_libpng]: A libpng installation already exists. -[install_libpng]: Delete "C:\carla\Build\libpng-1.2.37-install" if you want to force a rebuild. -[install_libpng]: Exiting... -[Setup]: Installing rpclib... -[install_rpclib]: [Batch params]: --build-dir "C:\carla\Build" -[install_rpclib]: A rpclib installation already exists. -[install_rpclib]: Delete "C:\carla\Build\rpclib-install" if you want to force a rebuild. -[install_rpclib]: Exiting... -[Setup]: Installing Google Test... -[install_gtest]: [Batch params]: --build-dir "C:\carla\Build" -[install_gtest]: A Google Test installation already exists. -[install_gtest]: Delete "C:\carla\Build\gtest-install" if you want to force a rebuild. -[install_gtest]: Exiting... -[Setup]: Installing "Recast & Detour"... -[install_recast]: [Batch params]: --build-dir "C:\carla\Build" -[install_recast]: A "Recast & Detour" installation already exists. -[install_recast]: Delete "C:\carla\Build\recast-0b13b0-install" if you want to force a rebuild. -[install_recast]: Exiting... -[Setup]: Installing Boost... -[install_boost]: [Batch params]: --build-dir "C:\carla\Build" --toolset msvc-14.2 --version 1.72.0 -j 16 -[install_boost]: A Boost installation already exists. -[install_boost]: Delete "C:\carla\Build\boost-1.72.0-install" if you want to force a rebuild. -[install_boost]: Exiting... 已复制 1 个文件。 已复制 1 个文件。 -[Setup]: Installing Xercesc... -[install_xercesc]: [Batch params]: --build-dir "C:\carla\Build" -[install_xercesc]: A xerces installation already exists. -[install_xercesc]: Delete "C:\carla\Build\xerces-c-3.2.3-install" if you want to force a rebuild. -[install_xercesc]: Exiting... 已复制 1 个文件。 -[Setup]: Installing Sqlite3 -[install_sqlite3]: [Batch params]: --build-dir "C:\carla\Build" -[install_sqlite3]: A Sqlite3 installation already exists. -[install_sqlite3]: Delete "C:\carla\Build\sqlite3-install" if you want to force a rebuild. -[install_sqlite3]: Exiting... 已复制 1 个文件。 -[Setup]: Installing PROJ -[install_proj]: [Batch params]: --build-dir "C:\carla\Build" -[install_proj]: A PROJ installation already exists. -[install_proj]: Delete "C:\carla\Build\proj-install" if you want to force a rebuild. -[install_proj]: Exiting... 已复制 1 个文件。 -[Setup]: Creating "CMakeLists.txt.in"... -[Setup]: ###########
> 
> # SUCCESS
> ###########
> 
> IMPORTANT!
> 
> All the CARLA library dependences should be installed now. (You can remove all "*-src" folders in C:\carla\Build\ directory)
> 
> You only need the ASSET PACK with all the meshes and textures.
> 
> This script provides the assets for CARLA 0.9.13: You can download the assets from here:
> 
> ```
>    http://carla-assets.s3.amazonaws.com/20211112_d5cfa12.tar.gz
> ```
> 
> Unzip it in the "C:\carla\Unreal\CarlaUE4\Content\Carla" folder. If you want another version, search it in C:\carla\Util\ContentVersions.txt.
> 
> -[BuildLibCarla]: [Batch params]: --server -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044. -- Build debug: ON -- Build release: ON -- Build test: ON -- Configuring done -- Generating done -- Build files have been written to: C:/carla/Build/libcarla-visualstudio 用于 .NET Framework 的 Microsoft (R) 生成引擎版本 16.11.2+f32259642 版权所有(C) Microsoft Corporation。保留所有权利。
> 
> carla_server.vcxproj -> C:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server.lib carla_server_debug.vcxproj -> C:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server_debug.lib -- Install configuration: "Release" -[BuildLibCarla]: LibCarla server has been successfully installed in "C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"! C:\carla\Util\BuildTools\Import.py:353: SyntaxWarning: "is not" with a literal. Did you mean "!="? if key is not 'tiles': Copying "C:\carla\Import\MapPackage\MyCity2\MyCity2.xodr" to "C:\carla\Util\DockerUtils\dist\MyCity2.xodr" Commandlet: ['D:\UnrealEngine/Engine/Binaries/Win64/UE4Editor', 'C:\carla\Unreal\CarlaUE4\CarlaUE4.uproject', '-run=ImportAssets', '-importSettings="C:\carla\importsetting.json"', '-nosourcecontrol', '-replaceexisting'] Copying "C:\carla\Import\MapPackage\MyCity2\MyCity2.fbx" to "C:\carla\Util\DockerUtils\dist\MyCity2.fbx" WARNING: cannot parse georeference: ''. Using default values. WARNING: Traffic sign 487 overlaps a driving lane. Moving out of the road... WARNING: Traffic sign 498 overlaps a driving lane. Moving out of the road... WARNING: Traffic sign 520 overlaps a driving lane. Moving out of the road... WARNING: Traffic sign 89 [... cut ...]
>   File "C:\Users\77550\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 373, in check_call
>   raise CalledProcessError(retcode, cmd)
>   subprocess.CalledProcessError: Command '['D:\UnrealEngine/Engine/Binaries/Win64/UE4Editor', 'C:\carla\Unreal\CarlaUE4\CarlaUE4.uproject', '-run=ImportAssets', '-importSettings="C:\carla\importsetting.json"', '-nosourcecontrol', '-replaceexisting']' returned non-zero exit status 3221225477.
>   make: *** [import] 错误 1

How did you export the map from carla ?

Blyron, 2026-11-12:
This is about the Unreal Engine 4 `make import` pipeline, which no longer exists. In CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) maps are imported with `Util/Tools/Import.py` or shipped as content packs, see [Custom maps](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_maps.md) and [Content packs](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_packs.md). Closing as obsolete. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron