carla-simulator/carla #5801: Strange neon lighting

kazimpal87, 2022-09-29:
Hi,

Sometimes when running carla, the lighting will go strange and produce images like this:

[image]

I don't really know why/when this happens.

I'm using carla 0.9.13 on Ubuntu.

XGodina, 2022-10-06:
This is because Vulkan runs out of memory and produces these effects.

kazimpal87, 2022-10-10:
why would that happen?

I am running a script which captures images from static cameras. Periodically the script loads a new map and spawns a load of new vehicles. If this is a memory thing, I guess either theres a memory leak or (more likely) I'm not properly cleaning up before loading the new map?

XGodina, 2022-10-17:
Yes, we always have memory problems with Vulkan, When loading to many assets usually happens this or directly crashes.

kazimpal87, 2022-10-18:
is there an alternative to Vulkan, or some other way to avoid this problem?

Blyron, 2024-05-03:
Hello!
We only support Vulkan in Ubuntu. Some users tried OpenGL, but we do not support that. Otherwise you could reduce your content quality.

https://carla.readthedocs.io/en/latest/adv_rendering_options/#setting-off-screen-mode-version-0912

Are you happy with this answer?

closed 2024-05-17 by Blyron