carla-simulator/carla #5806: Carla on WSL2 with GUI rendering

johnMinelli, 2022-10-03:
Hi I'm just filling this for the records if someone want to use Carla on WSL2 with GUI rendering.

This guide was very helpful to me for the audio setting:
https://carette.xyz/posts/setup_sound_video_wsl2/
For the video you need a custom `lavapipe`:
https://github.com/gnsmrky/wsl-vulkan-mesa
The readme guide says to set the DISPLAY setting at 1. That doesn't worked for me. I suggest instead:
- leave the setting in VcXsrv at -1
- tick "disable access control" in VcXsrv
- in WSL `export DISPLAY=$(ip route | awk '/^default/{print $3}'):0`
(as explained here https://github.com/gnsmrky/wsl-vulkan-mesa/issues/3)

(The prerequisite is a correct driver setup for your GPU in  windows to comunicate with WSL.)

I tested the behaviour just launching `./CarlaUE4.sh`
(if you don't want the renderings just add `-RenderOffscreen` and you are done)

---

**Troubleshooting**
- if you are seeing the black screen that opens and closes immediately with a segmentation fault like
```
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
```
check the DISPLAY env variable being setted in accordance with VcXsrv
- if you have a message of permission denied it's probably fauult of the Nvidia driver, as people suggest on internet. Someone says that strange resolutions on windows or not in accordance with the native one of the screen cause that, others solved by updating the drivers. In my case was having two screen. I disabled the secondary screen and all works fine.

CARLA version: 0.9.13
Platform/OS:  Windows 11, 22H2, Build: 22623.730
WSL2: Debian GNU/Linux 11 (bullseye)

I hope this saves someone else some time.

abidnazirisami, 2023-01-27:
Can you render CARLA using all of your GPU capabilities? I followed the links and I can get a render of CARLA from wsl, but it doesn't seem to be able to access the GPU in my system and is running very slow. I have the proper Nvidia drivers installed as well.

johnMinelli, 2023-01-27:
No, actually the last time I tried I had the same problem and I rollback to the Windows version 😓

ChanCody, 2023-08-01:
For anyone trying to use carla in wsl2, I suggest running the carla server in windows and running client in WSL. This works perfect for me.

sonalrpatel, 2023-09-19:
> For anyone trying to use carla in wsl2, I suggest running the carla server in windows and running client in WSL. This works perfect for me.

Could you please describe the steps to do it?

ChanCody, 2023-09-20:
1. Download the Carla server of Win version and start the Win server instead of Linux server.
2. In your client script, change the "localhost" to the ip of your Win.
3. Run your script in wsl and the client will connect to the Win server.

Mulberry26, 2023-10-25:
> 1. Download the Carla server of Win version and start the Win server instead of Linux server.
> 2. In your client script, change the "localhost" to the ip of your Win.
> 3. Run your script in wsl and the client will connect to the Win server.

connecting to Carla server...
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.6 
WARNING: Simulator API version  = baf43b07 

I've run into this problem. I was wondering if you have done anything else to fix this? Because Win doesn't have carla 9.6

angelomorgado, 2023-11-15:
> > 1. Download the Carla server of Win version and start the Win server instead of Linux server.
> > 2. In your client script, change the "localhost" to the ip of your Win.
> > 3. Run your script in wsl and the client will connect to the Win server.
> 
> connecting to Carla server... WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API WARNING: Client API version = 0.9.6 WARNING: Simulator API version = [baf43b0](https://github.com/carla-simulator/carla/commit/baf43b07a607e20bc338ff86d8829aae49f7478c)
> 
> I've run into this problem. I was wondering if you have done anything else to fix this? Because Win doesn't have carla 9.6

Hey last time I had a similar error it was because i needed to use Python 3.8, try running the client in a virtual env with conda

Sgt-Harry, 2024-04-01:
Do we still need a VcXsrv on Windows to use Carla on WSL2 when WSL2 now actually support GUI application?