carla-simulator/carla #5808: Wet Rainy Texture Render Error - Town10HD

jokolaco, 2022-10-03:
[image]

[image]

manuelmaior29, 2023-04-11:
Hello @jokolaco, I am experiencing the same issue. Did you manage to find any fix for this? Thanks!

ruff-huettinger, 2023-04-18:
Is there a way to get rid of this error in a package installation?

astralmaster, 2025-06-25:
Even with Gradient - Computational, I am still getting these flickers:

[image]

Blyron, 2026-11-12:
This was a rendering artefact of the Unreal Engine 4 materials. Rendering and weather materials are different in CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`). Closing as obsolete. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron