carla-simulator/carla #5814: Bug when spawning attached sensors using the ".then()" command

brscholz, 2022-10-06:
Hi Carla Devs,

I may have found a bug:

When I spawn a vehicle and in the same batch command want to spawn and attach a sensor using the ".then()" command, the attachment will be ignored and the sensor will be spawned with its transform relative to the scene origin.

`batch.append(carla.command.SpawnActor(actor_blueprint, spawn_point).then(carla.command.SpawnActor(camera_blueprint, carla.Transform(location=carla.Location(x=1,y=0,z=1.4),rotation=carla.Rotation()), carla.command.FutureActor)))`

Best regards,
Robert

SExpert12, 2023-07-04:
Hi,
Do you get the solution for this?
I also want to generated multiple vehicles and I wanted to attach sensor with each one.

qhaas, 2024-10-24:
The ability to do this would be useful for our efforts as well and doesn't seem to be supported in the head of the ue5-dev branch at time of writing.

M-Colley, 2026-10-26:
Draft PR #10052 proposes the `ue58-dev` correction for this exact SpawnActor.then(SpawnActor(..., FutureActor)) pattern. The server now resolves the child's explicit FutureActor parent to the newly spawned vehicle id, preserving ordinary explicit/omitted parents. Actual native command/MsgPack tests pass 7/7 on Windows and Linux with sanitizer review. A live vehicle/GNSS attachment and relative-offset smoke regression is included but not run without a matching simulator; full Unreal compilation remains outstanding. This is the same root as #5711.