carla-simulator/carla #5825: Assertion failed for SafeDistanceSensor windows build

inv-midhunbenny, 2022-10-12:
@germanros1987 @nsubiron @stavBodik @Axel1092  @bernatx @marcgpuig 
Running Carla 0.9.13 on Windows 10 build from source as per the documentation using .whl file `carla-0.9.13-cp37-cp37m-win_amd64`(https://carla.readthedocs.io/en/0.9.13/build_windows/#part-two-build-carla)
But after adding the codes for safe distance sensor from the GitHub code given by @nsubiron (https://gist.github.com/nsubiron/011fd1b9767cd441b1d8467dc11e00f9)

I complied the code using the make PythonAPI and then make launch.

SafeDistanceSensor class added to manual_control.py
```
class SafeDistanceSensor(object):
    def __init__(self, parent_actor):
        self.sensor = None
        self._parent = parent_actor
        world = self._parent.get_world()
        bp = world.get_blueprint_library().find('sensor.other.safe_distance')
        self.sensor = world.spawn_actor(bp, carla.Transform(), attach_to=self._parent)
        self.sensor.listen(lambda event: SafeDistanceSensor.callback(world, event))
    @staticmethod
    def callback(world, event):
        for actor_id in event:
            vehicle = world.get_actor(actor_id)
            print('Vehicle too close: %s' % vehicle.type_id)
```
Upon executing the code and driving my car to a near by vehicle the server crashes and assertation error is been logged.
`Assertion failed: px != 0, file C:\workspace\carla\Build\boost-1.72.0-install\include\boost/smart_ptr/shared_ptr.hpp, line 734`

myleskeller, 2023-02-28:
I'm experiencing this issue as well. I'm also on Win10, but am running 0.9.**14**.

All I can determine from researching the error is that something is trying to use an uninitialized shared pointer.

I know that in `SensorData.cpp`, the tutorial instructs us to add
```
    class_<csd::SafeDistanceEvent,                    // actual type.
           bases<cs::SensorData>,                     // parent type.
           boost::noncopyable,                        // disable copy.
           boost::shared_ptr<csd::SafeDistanceEvent>  // use as shared_ptr.
           >("SafeDistanceEvent", no_init)            // name, and disable construction.
        .def("__len__", &csd::SafeDistanceEvent::size)
        .def("__iter__", iterator<csd::SafeDistanceEvent>())
        .def(
            "__getitem__", +[](const csd::SafeDistanceEvent &self, size_t pos) -> cr::ActorId { return self.at(pos); });
```
which makes me wonder if it is rooted in how/when the class is initialized because of this part of the declaration:
```
boost::shared_ptr<csd::SafeDistanceEvent>  // use as shared_ptr.
```

Blyron, 2026-11-12:
The sensor tutorial has been rewritten for Unreal Engine 5, see [Create a sensor](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/dev_create_sensor.md). Closing. If it still asserts with CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`), please open a new issue with the stack trace.

closed 2026-11-12 by Blyron