carla-simulator/carla #5827: How to make the motorcycle lane splitting?

FrankZTH, 2022-10-13:
Hello, everyone,
I need to make Carla close to Asia traffic conditions.
Like the picture below.
[image]

But now I have no idea how to make it work.
Is this work could make by PythonAPI?
Any help will be appreciated !!!

CARLA version: 0.9.13
Platform/OS: Linux 18.04

TR-Monash, 2022-10-18:
You can perhaps use the `offset` parameter when you setup the agent. For example, something like below:
`cycle_agent = BasicAgent(bike_vehicle, opt_dict={'offset': 1.3, 'ignore_vehicles': True})`
However, a larger (or smaller) `offset `may break the autopilot algorithm.

FrankZTH, 2022-10-24:
> You can perhaps use the `offset` parameter when you set up the agent. For example, something like below: `cycle_agent = BasicAgent(bike_vehicle, opt_dict={'offset': 1.3, 'ignore_vehicles': True})` However, a larger (or smaller) `offset `may break the autopilot algorithm.

Thanks for your answer
I have already tried this method, but it seems not to work.
I'll keep trying this method. 
I appreciate any answers.