carla-simulator/carla #5842: Instance Segmentation - Standard semantic ID only contains subset of objects

jokolaco, 2022-10-17:
According to the documentation " The R channel contains the standard semantic ID."
While this is true and working in the semantic segmentation case i noticed in the instance segmentation case
that e.g. cars are in the red channels 9,10,11 instead of only the 10th redchannel.

This makes it hard to access the segmented areas because they are not clearly living in one specific channel.
This is also the case for the "actor id" tagging.

Input:
[image]
Mask:
[image]

Redschannel:
9
[image]
10
[image]
11
[image]

CARLA version:
9.13 and latest night build, standard downloads from github and self compiled on Rocky Linux 8

Platform/OS:
Ubuntu 18 and Rocky Linux 8

mozg1, 2023-02-16:
Can the Instance Segmentation sensor finally be looked at for the next release, because it is simply not working according to the documentation. My thesis depends on this, thanks.

JosenJiang, 2023-03-02:
Do you save instance segmentation image as JPG format? Please save it as PNG format.

CaphalorAlb, 2023-11-07:
> Do you save instance segmentation image as JPG format? Please save it as PNG format. JPG is a lossy compression format, and some pixel's value will change when you save images as xx.jpg. PNG is lossless compression format.

hey, I'm not sure how to do this properly, running into the same issue

this is what i'm using

`image_pv_seg.save_to_disk(frame_path_seg + '.png')`

CaphalorAlb, 2023-11-13:
> According to the documentation " The R channel contains the standard semantic ID." While this is true and working in the semantic segmentation case i noticed in the instance segmentation case that e.g. cars are in the red channels 9,10,11 instead of only the 10th redchannel.
> 
> This makes it hard to access the segmented areas because they are not clearly living in one specific channel. This is also the case for the "actor id" tagging.
> 
> Input: [image] Mask: [image]
> 
> Redchannel: 9 [image] 10 [image] 11 [image]
> 
> CARLA version: 9.13 and latest night build, standard downloads from github and self compiled on Rocky Linux 8
> 
> Platform/OS: Ubuntu 18 and Rocky Linux 8

did you ever find a solution for this? running into the same issue and i'm not sure how to work around it

Blyron, 2024-11-06:
Due to a contribution from @AreopagX Now the actor id generated by carla matches the instance segmentation one;
For docs check; https://carla.readthedocs.io/en/latest/ref_sensors/#instance-segmentation-camera
https://carla.readthedocs.io/en/latest/tuto_G_instance_segmentation_sensor/
https://carla.readthedocs.io/en/latest/tuto_D_create_semantic_tags/

Usually the id in red channel is Unreal generated actor id which differs from CARLA's.  that's why everyobject can be shown in the red channel.