carla-simulator/carla #5846: The download website of Xerces-c 3.2.3 is not aviliable now

Nash-dc, 2022-10-18:
Recently, when I run make PythonAPI, I encountered the following problem:

404 Not Found in https://ftp.cixug.es/apache//xerces/c/3/sources/xerces-c-3.2.3.tar.gz

It looks like I can't download xerces-c-3.2.3.tar.gz from ftp.cixug.es
then I tried to open https://ftp.cixug.es/apache//xerces/c/3/sources
then I found there is only xerces-c-3.2.4,no xerces-c-3.2.3.

Try to solve it, I replace line 432 in /Util/BuilldTools/Setup.sh with XERCESC_REPO=https://archive.apache.org/dist/xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz
then the problem is temporarily solved,and the build is sucessful.But I am not sure whether this operation is correct.Can someone help me?

CARLA version:0.9.13
Platform/OS:Ubuntu 20.04
Problem I have experienced:The download website of Xerces-c 3.2.3 is not aviliable

keeees, 2022-10-18:
Same issue here

Nash-dc, 2022-10-18:
> Same issue here

try my solution,it works

keeees, 2022-10-18:
Yeah it works! Thank you

sebastienmascha, 2022-10-23:
It is already fixed in `dev`: https://github.com/carla-simulator/carla/blob/dev/Util/BuildTools/Setup.sh

olyii, 2022-11-02:
doesn't work,why?

aamzhas, 2022-11-04:
FYI I don't think this is reflected in the release version. Building this via Docker will still give the same error unless explicitly replaced after pull from git.

kyle-sama, 2022-11-07:
Changing the address in setup.sh doesn't seem to solve the issue if you already built the prequisites docker... seems like it caches the setup script somewhere.

zehranrgi, 2023-06-15:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

Yes, it has still given the same problem. But the solution provided by @Nash-dc worked.

Aestial, 2023-06-27:
Same issue building Carla 0.9.13 What I did was to download xerces-c source (currently v3.2.4) and place its contents inside carla/Build/xerces-c-3.2.3-source directory and it seems to work. Only I got several warnings because kind of detected the newer version.

HSqure, 2023-08-07:
It works on my envirment with ubuntu 18.04! Thanks!

closed 2024-05-10 by Blyron