carla-simulator/carla #5847: Clang 8 is required, but is not installed while compiling

mgmike, 2022-10-19:
Thanks for contributing to CARLA!

If you are reporting an issue, please use the following outline:

CARLA version: 0.9.13 (from source https://carla.readthedocs.io/en/latest/build_linux/)
Platform/OS: Ubuntu 20.04

Problem: 
When compiling PythonAPI, I get error "clang 8 is required, but it's not installed."

Expectation: 
The [installation doc](https://carla.readthedocs.io/en/latest/build_linux/) mentions
> The CARLA team uses clang-8 (or clang-10 in Ubuntu 20.04)

So I shouldn't need to install clang-8. 

Steps to reproduce:
I have followed the steps taken in the [installation doc](https://carla.readthedocs.io/en/latest/build_linux/) for Ubuntu 20.04. My update-alternatives is set to 10 for both clang++ and clang. I also confirmed that the Unreal Engine was built using clang-10:

> ------- Build details --------
Using toolchain located at '/home/mike/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/mike/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '10.0.1' (string), 10 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /home/mike/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).

jokolaco, 2022-10-19:
It is easy to solve this by investigation of the error message and the origin of the message.
Open Setup.sh and search for "clang 8 is required, but it's not installed." 
Now you can read that the script exports some CC and CXX settings. Obviously you have not installed clang 8 but as you possibly can read the system tries according to the used script to use clang 8.

Now you could just install clang 8 OR change the setting in the script to e.g. "clang++-10" and "clang-10".

Moneyboy63211, 2022-10-21:
> It is easy to solve this by investigation of the error message and the origin of the message. Open Setup.sh and search for "clang 8 is required, but it's not installed." Now you can read that the script exports some CC and CXX settings. Obviously you have not installed clang 8 but as you possibly can read the system tries according to the used script to use clang 8.
> 
> Now you could just install clang 8 OR change the setting in the script to e.g. "clang++-10" and "clang-10".
> 
> It is a little bit confusing, but as far as i remember the mentioned toolchain is used to compile the Unreal Engine but the carla stuff is build using the system settings. So you have to take care that your environment has all the dependencies on board. And if not adjust the build scripts.

Hi , 

Hello ,

thanks for your help.
But i followed your way to solve this problem and i got another problem.

I got the following error when "make PythonAPI":

"clang++" -c -x c++ -fvisibility-inlines-hidden -fPIC -m64 -pthread -O3 -Wall -fvisibility=hidden -Wno-inline -fPIC -std=c++14 -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_ALL_NO_LIB=1 -DBOOST_NUMPY_SOURCE -DNDEBUG -I"." -I"/home/tianle/venv_carla/.venv/include/python3.7" -I"/home/tianle/venv_carla/.venv/lib/python3.7/site-packages/numpy/core/include" -o "bin.v2/libs/python/build/clang-linux-10.0/release/python-3.7/threading-multi/visibility-hidden/numpy/ndarray.o" "libs/python/src/numpy/ndarray.cpp"

...failed clang-linux.compile.c++.without-pch bin.v2/libs/python/build/clang-linux-10.0/release/python-3.7/threading-multi/visibility-hidden/numpy/ndarray.o...
clang-linux.compile.c++.without-pch bin.v2/libs/python/build/clang-linux-10.0/release/python-3.7/threading-multi/visibility-hidden/numpy/ufunc.o
In file included from libs/python/src/numpy/ufunc.cpp:8:
In file included from ./boost/python/numpy/internal.hpp:17:
In file included from ./boost/python.hpp:11:
In file included from ./boost/python/args.hpp:8:
In file included from ./boost/python/detail/prefix.hpp:13:
./boost/python/detail/wrap_python.hpp:57:11: fatal error: 'pyconfig.h' file not found
# include <pyconfig.h>
          ^~~~~~~~~~~~
1 error generated.

  "clang++" -c -x c++ -fvisibility-inlines-hidden -fPIC -m64 -pthread -O3 -Wall -fvisibility=hidden -Wno-inline -fPIC -std=c++14 -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_ALL_NO_LIB=1 -DBOOST_NUMPY_SOURCE -DNDEBUG -I"." -I"/home/tianle/venv_carla/.venv/include/python3.7" -I"/home/tianle/venv_carla/.venv/lib/python3.7/site-packages/numpy/core/include" -o "bin.v2/libs/python/build/clang-linux-10.0/release/python-3.7/threading-multi/visibility-hidden/numpy/ufunc.o" "libs/python/src/numpy/ufunc.cpp"

...failed clang-linux.compile.c++.without-pch bin.v2/libs/python/build/clang-linux-10.0/release/python-3.7/threading-multi/visibility-hidden/numpy/ufunc.o...
clang-linux.compile.c++.without-pch bin.v2/libs/python/build/clang-linux-10.0/release/python-3.7/threading-multi/visibility-hidden/numpy/scalars.o
In file included from libs/python/src/numpy/scalars.cpp:8:
In file included from ./boost/python/numpy/internal.hpp:17:
In file included from ./boost/python.hpp:11:
In file included from ./boost/python/args.hpp:8:
In file included from ./boost/python/detail/prefix.hpp:13:
./boost/python/detail/wrap_python.hpp:57:11: fatal error: 'pyconfig.h' file not found
# include <pyconfig.h>
          ^~~~~~~~~~~~
1 error generated.

  "clang++" -c -x c++ -fvisibility-inlines-hidden -fPIC -m64 -pthread -O3 -Wall -fvisibility=hidden -Wno-inline -fPIC -std=c++14 -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_ALL_NO_LIB=1 -DBOOST_NUMPY_SOURCE -DNDEBUG -I"." -I"/home/tianle/venv_carla/.venv/include/python3.7" -I"/home/tianle/venv_carla/.venv/lib/python3.7/site-packages/numpy/core/include" -o "bin.v2/libs/python/build/clang-linux-10.0/release/python-3.7/threading-multi/visibility-hidden/numpy/scalars.o" "libs/python/src/numpy/scalars.cpp"

...failed clang-linux.compile.c++.without-pch bin.v2/libs/python/build/clang-linux-10.0/release/python-3.7/threading-multi/visibility-hidden/numpy/scalars.o...
clang-linux.compile.c++.without-pch bin.v2/libs/python/build/clang-linux-10.0/release/python-3.7/threading-multi/visibility-hidden/numpy/matrix.o
In file included from libs/python/src/numpy/matrix.cpp:8:
In file included from ./boost/python/numpy/internal.hpp:17:
In file included from ./boost/python.hpp:11:
In file included from ./boost/python/args.hpp:8:
In file included from ./boost/python/detail/prefix.hpp:13:
./boost/python/detail/wrap_python.hpp:57:11: fatal error: 'pyconfig.h' file not found
# include <pyconfig.h>
          ^~~~~~~~~~~~
1 error generated.

..
..
..
.
.

ons-variant-shared.cmake
common.copy /home/tianle/Documents/carla/Build/boost-1.72.0-c8-source/stage/lib/libboost_program_options.so.1.72.0
ln-UNIX /home/tianle/Documents/carla/Build/boost-1.72.0-c8-source/stage/lib/libboost_program_options.so
ln-UNIX /home/tianle/Documents/carla/Build/boost-1.72.0-c8-source/stage/lib/libboost_program_options.so.1
ln-UNIX /home/tianle/Documents/carla/Build/boost-1.72.0-c8-source/stage/lib/libboost_program_options.so.1.72
common.copy /home/tianle/Documents/carla/Build/boost-1.72.0-c8-source/stage/lib/cmake/boost_program_options-1.72.0/libboost_program_options-variant-shared.cmake
...failed updating 66 targets...
...skipped 24 targets...
...updated 173 targets...
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1

mgmike, 2022-10-21:
> It is easy to solve this by investigation of the error message and the origin of the message. Open Setup.sh and search for "clang 8 is required, but it's not installed." Now you can read that the [...]

@jokolaco I changed all the references of clang-8 and clang++-8 to 10. I had to change these in BuildOSM2ODR.sh and BuildPythonAPI.sh as well and the build works. I will create a branch soon with these updates.

mgmike, 2022-10-21:
@Moneyboy63211 you will have to change BuildOSM2ODR.sh and BuildPythonAPI.sh if you have not already. You may also run into [this issue](https://github.com/carla-simulator/carla/issues/2416). 

I think the solution to your issue, if you are also using ubuntu 20 though, is to specify python3.8 when you build 
`make PythonAPI ARGS="--python-version=3.8"`

awiswasi, 2023-07-25:
> @Moneyboy63211 you will have to change BuildOSM2ODR.sh and BuildPythonAPI.sh if you have not already. You may also run into [this issue](https://github.com/carla-simulator/carla/issues/2416).
>
> I [...]

Could you please tell me what needs to be changed?
I changed all instances of `clang-8` to `clang-10`.
`make setup` and `make LibCarla` both finished successfully, however when I `make PythonAPI`I get this error:

`carla-0.9.13/PythonAPI/carla/dependencies/lib/libproj.a(geodesic.c.o): relocation R_X86_64_32S against .rodata.cst8' can not be used when making a shared object; recompile with -fPIC /home/t/anaconda3/compiler_compat/ld: /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libproj.a(wkt1_generated_parser.c.o): relocation R_X86_64_32S against .rodata' can not be used when making a shared object; recompile with -fPIC`

IDontEatFish, 2023-08-24:
`sudo apt-get install build-essential clang-8 lld-8 g++-7 cmake ninja-build libvulkan1 python python-pip python-dev python3-dev python3-pip libpng-dev libtiff5-dev libjpeg-dev tzdata sed curl unzip autoconf libtool rsync libxml2-dev git`

Blyron, 2026-11-12:
This is an error from the old Make/`Setup.sh` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake and uses the Unreal Engine toolchain. See [Build on Linux](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_linux_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron