carla-simulator/carla #5852: Impossible to build the dependency xerces 3.2.3 since they upgraded to 3.2.4

sebastienmascha, 2022-10-20:
CARLA version: 0.9.13
Platform/OS: Ubuntu 18
Problem you have experienced: Impossible to build the dependency xerces 3.2.3
What you expected to happen: 
- Cannot download the archived 3.2.3 version: https://ftp.cixug.es/apache/xerces/c/3/sources/xerces-c-3.2.3.tar.gz
- This file doesn't exist since they released a new version on October 17th 2022
- We should migrate to 3.2.4: https://ftp.cixug.es/apache/xerces/c/3/sources/xerces-c-3.2.4.tar.gz
Steps to reproduce: Build the Carla.Dockerfile using master or 0.9.13 branches.
Other information (documentation you consulted, workarounds you tried): https://ftp.cixug.es/apache/xerces/c/3/sources/

sebastienmascha, 2022-10-22:
It has been fixed in the latest develop. Thanks!

closed 2022-10-22 by sebastienmascha

hexrefugee, 2023-01-06:
3q