carla-simulator/carla #5857: Gear inversion can be operated regardless of the vehicle speed

lucalazzaroni, 2022-10-21:
Hi everyone,
I am currently experiencing some issues with the gear inversion in Carla. I noticed through the manual control that the reverse gear can be activated (Q key) regardless of the current speed of the vehicle. I would like to change this behavior since it is not conform to the real one. More in detail, I would like to set the reverse flag to be actionable only when the the vehicle is stopped or it has a really low speed.
I could patch it up from the client side (by setting certain boundaries before applying the ```apply_control``` function), but would it be possible to implement this directly from UE4?

Thanks in advance to those who would like to help me out.

CARLA version: 0.9.13
Platform/OS: Windows 11/Ubuntu 20.04
Problem you have experienced:
What you expected to happen:
Steps to reproduce:
Other information (documentation you consulted, workarounds you tried):

lucalazzaroni, 2022-11-03:
Anyone?

Blyron, 2022-11-21:
Hello Luca! 

I have been looking at it. Currently is difficult to us to implement it directly in UE4 as we have to connect lights system and gear system inside engine. We will try to make it in the future, 

Thank you for your time and post!