carla-simulator/carla #5870: make: *** [Makefile:327: CrashReportClient-Linux-Shipping] Error 6

Alphabaijinde, 2022-10-26:
**Platform is the Ubuntu20.04, UE4 is the carla UE4 branch carla, carla is the 0.9.13** 

./GenerateProjectFiles.sh
The following is the specific operation information:

Attempting to set up UE4 pretty printers for gdb (existing UE4Printers.py, if any, will be overwritten)...
        updated UE4Printers.py
        cannot modify .gdbinit. Please add the below lines manually:

python
        import sys
        sys.path.append('/home/baijinde/.config/Epic/GDBPrinters/')
        from UE4Printers import register_ue4_printers
        register_ue4_printers(None)
        print("Registered pretty printers for UE4 classes")
        end

Setting up Unreal Engine 4 project files...

Fixing inconsistent case in filenames.
Setting up Mono
Generating data for project indexing...   
WARNING: Exception while generating include data for UE4Editor: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for BenchmarkTool: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for BlankProgram: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for BuildPatchTool: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for CrashReportClient: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for CrashReportClientEditor: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for DatasmithSDK: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for HeadlessChaos: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for LiveCodingConsole: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for ShaderCompileWorker: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for SlateViewer: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for SwitchboardListener: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for TestPAL: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for TextureShareSDK: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealCEFSubProcess: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealFileServer: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealFrontend: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealHeaderTool: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealInsights: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealLightmass: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealMultiUserServer: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealPak: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealRecoverySvc: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealVersionSelector: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for BootstrapPackagedGame: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
Generating data for project indexing... 100%
Generating data for project indexing...   
WARNING: Exception while generating include data for UE4Editor: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for BenchmarkTool: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for BlankProgram: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for BuildPatchTool: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for CrashReportClient: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for CrashReportClientEditor: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for DatasmithSDK: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for HeadlessChaos: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for LiveCodingConsole: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for ShaderCompileWorker: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for SlateViewer: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for SwitchboardListener: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for TestPAL: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for TextureShareSDK: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealCEFSubProcess: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealFileServer: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealFrontend: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealHeaderTool: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealInsights: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealLightmass: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
WARNING: Exception while generating include data for UnrealMultiUserServer: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
[... 101 lines cut ...]
WARNING: Exception while generating include data for TextureShareSDK: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
[... cut ...]
Fixing inconsistent case in filenames.
Setting up Mono
Running command : Engine/Binaries/DotNET/UnrealBuildTool.exe CrashReportClient Linux Shipping
Using 'git status' to determine working set for adaptive non-unity build (/home/baijinde/work/data/github/UnrealEngine_4.26).
Creating makefile for CrashReportClient (no existing makefile)
ERROR: Platform Linux is not a valid platform to build. Check that the SDK is installed properly.
make: *** [Makefile:327: CrashReportClient-Linux-Shipping] Error 6

whan2013xh, 2022-11-10:
I have the same problem。
Solution:
1、download the toolchains.choose your UE version,my version is 4.26.
https://docs.unrealengine.com/4.26/en-US/SharingAndReleasing/Linux/NativeToolchain/

2、tar the file,then overwrite the path /home/username/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/ files.

AndreaCatania, 2022-12-06:
@whan2013xh Many thanks for you input, however I've tried what you suggested but it didn't change much. I still have the exact same issue :thinking: 

I'm on Fedora 37, Unreal version 4.27

Alphabaijinde, 2022-12-06:
这是来自QQ邮箱的假期自动回复邮件。您好，我已经收到您的邮件，会尽快回复的。

wcole3, 2023-02-08:
Just hit this and figured I'd add a data point.  I was able to solve the issue by manually downloading the toolset as suggested by @whan2013xh above; interestingly we only encountered this on 20.04 VM's and 22.04 VM's built without issue.

huangchunan, 2024-04-18:
> @whan2013xh Many thanks for you input, however I've tried what you suggested but it didn't change much. I still have the exact same issue 🤔
> 
> I'm on Fedora 37, Unreal version 4.27

The same. Have u solved this？

huangchunan, 2024-04-18:
The same. Have u solved this？

> @whan2013xh Many thanks for you input, however I've tried what you suggested but it didn't change much. I still have the exact same issue 🤔
> 
> I'm on Fedora 37, Unreal version 4.27

The same. Have u solved this？

mobiiin, 2024-08-12:
I'm still having the same error even though I've downloaded the toolchain as instructed. 
Ubuntu 22
UE 5.3 
any help is appreciated

Blyron, 2026-11-12:
This is an error from the old Make/`Setup.sh` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake and uses the Unreal Engine toolchain. See [Build on Linux](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_linux_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron