carla-simulator/carla #5872: Dormant/Collided vehicles disappear even after commenting lines in ALSM.CPP

sk9073, 2022-10-26:
Version: 0.9.13
Platform: Windows 10

Setting: I spawn something like 100 vehicles using the PythonAPI/examples/generate_traffic.py script by making each vehicle ignore traffic_light 100% of the time to create traffic congestion,
and set up another script that polls the locations of all actors in the simulation and writes them to a file

Issue: The actors that remain dormant/have collided with other actors get destroyed after a min.
I added this argument to trafficManager but this didn't have any effect : 
           `traffic_manager.set_respawn_dormant_vehicles(False)`

Following the steps of #3860  I even commented on these lines in ALSM.CPP but it doesn't seem to have any effect on Lines (106-111).

          // if (IsVehicleStuck(max_idle_time.first)
          //     && (current_timestamp.elapsed_seconds - elapsed_last_actor_destruction) > DELTA_TIME_BETWEEN_DESTRUCTIONS) {
          //   registered_vehicles.Destroy(max_idle_time.first);
          //   RemoveActor(max_idle_time.first, true);
          //   elapsed_last_actor_destruction = current_timestamp.elapsed_seconds;
          // }

SExpert12, 2025-04-08:
Hi, have you resolve this issue?
I am facing the same problem.

sk9073, 2025-04-08:
No @SExpert12 , I just had to run the simulator on a pc with btr GPU.

SExpert12, 2025-04-08:
okay. thanks for reply.