carla-simulator/carla #5883: ERROR: No matching distribution found for carla==0.9.11

molua, 2022-10-27:
Platform/OS: windows10
Problem you have experienced: when I wanner download carla 0.9.11 package in command by`pip install carla==0.9.11` 
I got 
```
ERROR: Could not find a version that satisfies the requirement carla==0.9.11 (from versions: 0.9.5, 0.9.12, 0.9.13)
ERROR: No matching distribution found for carla==0.9.11
```
How do I download the corespond version 0.9.11 carla package

imsrinin, 2022-10-31:
I am having the same issue for a while now. OS: Ubuntu 20.04
command: pip install carla==0.9.13

ERROR: Could not find a version that satisfies the requirement carla==0.9.13 (from versions: 0.9.5)
ERROR: No matching distribution found for carla==0.9.13

fawadshahid, 2022-11-03:
> Platform/OS: windows10 Problem you have experienced: when I wanner download carla 0.9.11 package in command by`pip install carla==0.9.11` I got
>
> ```
> ERROR: Could not find a version that satisfies [...]

I dont think Carla 0.9.11 is on pip you have to use the .egg for 0.9.11

fawadshahid, 2022-11-03:
> I am having the same issue for a while now. OS: Ubuntu 20.04 command: pip install carla==0.9.13
> 
> ERROR: Could not find a version that satisfies the requirement carla==0.9.13 (from versions: 0.9.5) ERROR: No matching distribution found for carla==0.9.13

Check your pip version.

yaoweixiao-ux, 2023-02-13:
ubuntu18.04 have carla==0.9.13

zillur-av, 2023-02-23:
You can install carla client using pip if it is carla 0.9.12+. Just run ```pip3 install --upgrade pip``` and  ```pip install carla```

zillur-av, 2023-03-01:
> > You can install carla client using pip if it is carla 0.9.12+. Just run `pip3 install --upgrade pip` and `pip install carla`
> 
> I'm facing this problem and I don't know how to solve it because I use Simulator API version = 0.9.11 and with pip it provide only versions: 0.9.5, 0.9.12, 0.9.13, 0.9.14.

In that case, you do not need to do anything according to [docs](https://carla.readthedocs.io/en/0.9.11/start_quickstart/#carla-installation). Your installation should come with the required .egg file in ```./PythonAPI/carla/dist/```.

skyetheguy, 2024-05-08:
I'm having a similar issue trying to install carla client lib in an ros2 docker container (osrf/ros:foxy-desktop).
command:
`pip install carla==0.9.13`
result:
```
ERROR: Could not find a version that satisfies the requirement carla==0.9.13 (from versions: 0.9.5)
ERROR: No matching distribution found for carla==0.9.13
```

Trying to install the whl from /PythonAPI/carla/dist/
command:
`pip install carla-0.9.13-cp37-cp37m-manylinux_2_27_x86_64.whl`
result:
`ERROR: carla-0.9.13-cp37-cp37m-manylinux_2_27_x86_64.whl is not a supported wheel on this platform.`

yizh7963, 2024-06-02:
> I'm having a similar issue trying to install carla client lib in an ros2 docker container (osrf/ros:foxy-desktop). command: `pip install carla==0.9.13` result:
>
> ```
> ERROR: Could not find a version [...]

I have the similar situation and I guess the python version is the cause (but I am not sure of it)
I have python=3.11 installed by conda, and this errors occur when I pip install carla==0.9.15
but when I create a new env python=3.10, no error occurs.

privateboss0, 2024-11-03:
I ran into this issue also, but a thing to note is the Api's might have been natively integrated if you get the error and you don't need to install Carla using pip. In my case I just went on to spawn_NPC.py or dynamic_weather.py. using conda virtual python environment when I got the error too and it works :)

Blyron, 2026-11-12:
Old versions were never published on PyPI. Install the wheel shipped in the package under `PythonAPI/carla/dist`, see [Quick start](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/start_quickstart.md). Closing.

closed 2026-11-12 by Blyron