carla-simulator/carla #5886: Build error in Setup.sh: Compiling libc++

mgmike, 2022-10-29:
CARLA version: 0.9.13 dev and master
Platform/OS: Ubuntu 20.04
Problem you have experienced: build error on Setup.sh Compiling libc++
What you expected to happen: Build with no error
Steps to reproduce: Fresh install on Ubuntu 20.04 and follow [the Carla install from source docs](https://carla.readthedocs.io/en/latest/build_linux/) 
Other information:

In the master branch I changed all mentions of clang-8 and clang++-8 to 10 in Setup.sh, BuildPythonAPI.sh and BuildOSM2ODR.sh. For both that and dev I get the following error

``` 
Setup.sh: Compiling libc++.
-- The C compiler identification is Clang 10.0.0
-- The CXX compiler identification is Clang 10.0.0
-- The ASM compiler identification is Clang
-- Found assembler: /usr/bin/clang-10
-- Check for working C compiler: /usr/bin/clang-10
-- Check for working C compiler: /usr/bin/clang-10 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-10
-- Check for working CXX compiler: /usr/bin/clang++-10 -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  The C++ compiler

    "/usr/bin/clang++-10"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/mike/carla/Build/llvm-8.0-build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/ninja cmTC_d50bb && [1/2] Building CXX object CMakeFiles/cmTC_d50bb.dir/testCXXCompiler.cxx.o
    [2/2] Linking CXX executable cmTC_d50bb
    FAILED: cmTC_d50bb 
    : && /usr/bin/clang++-10     CMakeFiles/cmTC_d50bb.dir/testCXXCompiler.cxx.o  -o cmTC_d50bb   && :
    /usr/bin/ld: cannot find -lstdc++
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:39 (project)

-- Configuring incomplete, errors occurred!
See also "/home/mike/carla/Build/llvm-8.0-build/CMakeFiles/CMakeOutput.log".
See also "/home/mike/carla/Build/llvm-8.0-build/CMakeFiles/CMakeError.log".
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1
```

Seems like there is an issue with clang being 10 and llvm being 8. I tried changing LLVM_BASENAME to llvm-10.0 on line 61 of Util/BuildTools/Setup.sh but got the same error. I used a completely clean install of Ubuntu 20.04 after other issues. 
Any help would be appreciated.

mgmike, 2022-11-01:
I fixed this issue by adding a symbolic link
`sudo ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 /usr/lib/libstdc++.so`

But now I get an issue when compiling libc++ during "Performing test LLVM_LIBSTDCXX_MIN"
```
CMake Error at cmake/modules/CheckCompilerVersion.cmake:72 (message):
  libstdc++ version must be at least 4.8.
Call Stack (most recent call first):
  cmake/config-ix.cmake:13 (include)
  CMakeLists.txt:590 (include)
```

I am going to try to manually build this

Dragonking3x, 2022-11-10:
Try to use the requirements for Ubuntu 18.04
```
sudo apt-get install build-essential clang-8 lld-8 g++-7 cmake ninja-build libvulkan1 python3-dev python3-pip libpng-dev libtiff5-dev libjpeg-dev tzdata sed curl unzip autoconf libtool rsync libxml2-dev git
```
```
sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 &&
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180
```

maybe it helps to add this line to your .bashrc file
```
export CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH:/usr/bin/python3.8
```

wutianze, 2022-12-07:
This happens because the files in Util/BuildTools still use clang8, replace clang 8 with clang 10 manually. Also llvm 8 with llvm 10

leeseho, 2023-01-26:
I had the same error and solve this.

refer to

https://stackoverflow.com/questions/74543715/usr-bin-ld-cannot-find-lstdc-no-such-file-or-directory-on-running-flutte/74605488#74605488?newreg=27e5d25cd2c741b99cec030fde85633d

and you don't have to  change 'LLVM_BASENAME'

Blyron, 2026-11-12:
This is an error from the old Make/`Setup.sh` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake and uses the Unreal Engine toolchain. See [Build on Linux](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_linux_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron