carla-simulator/carla #5887: simulation time of replay

Taekbum, 2022-10-29:
I recorded scenario and played the scenario several times with different environment (weather, etc..). I tried to get synchronized video from each recorded scenario. To do so, I used frame value of received data from sensor (every time frame value % 10 is increased, save image) so that I can save exactly same scene if recorded file uses same simulation time as original scenario and timestamp of data is simulation time. However, timestamp values of images saved from each replays were not synchronized. 
What is the reason? Is it because image.timestamp != simulation time, or, simulation time of replay is not same as original scenario?

lowbees, 2022-11-01:
Simulation time is not replay time, replay time needs to be calculated

Fengzexu, 2023-02-06:
@lowbees ,hello, do you know how to calculate the replay time?

Blyron, 2026-11-12:
Sensor timestamps are simulation time. A replay reproduces the timing of the recording only when the server replays in synchronous mode with the same `fixed_delta_seconds`; match data by frame, not wall time. See [Recorder](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/core_recorder.md). Closing.

closed 2026-11-12 by Blyron