carla-simulator/carla #5889: HOW TO INSTALL THIS THING ON WINDOWS

excep, 2022-10-29:
Hello, 
 I followed this article here : https://carla.readthedocs.io/en/0.8.4/how_to_build_on_windows/ in order to install and run CARLA on windows, but apparently it's outdated. because there's no setup.bat file in the repository. so I was wondering if there's an alternative to install it.

excep, 2022-10-30:
all links are broken or don't exist anymore !!

jokolaco, 2022-10-31:
You can signup and post on github but you cannot browse the web? You ask how to build the latest carla version 0.9.13 and post an outdated documentation from version 0.8.4.

It is always advisable to read the latest information :
https://carla.readthedocs.io/en/latest/build_windows/https://carla.readthedocs.io/en/latest/build_windows/

excep, 2022-10-31:
I tried man I tried and I saw this link and followed instructions but when I get to where I should clone UnrealEngine repository I get the following Error and I don't know what is wrong : 
I enter command : 
 git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git .
 
 then a request to login to my Github accoutn pops up. I give it access via Token. then the following message shows up : 
 
 Cloning into 'UnrealEngine'...
remote: Repository not found.
fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found

jokolaco, 2022-11-01:
Oh gosh, 
yes it is complicated to compile unreal engine and carla from scratch. 
So you should read carefully the docs and spend some DAYS to understand everything.
E.g. you must register to Unreal in order to get access to the Unreal Engine...

excep, 2022-11-01:
I finaly found the waaaaaay Thaaank you Joko loco !!

closed 2023-02-01 by glopezdiest

northbigpanda, 2023-08-01:
> I finaly found the waaaaaay Thaaank you Joko loco !!

i also encounter the same issue, and i also have linked the account betwenn UE and github, but issue still exsited:
Cloning into 'UnrealEngine'...
remote: Repository not found.
fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found

how do you resolve this issue, thank you very much.