carla-simulator/carla #5891: Attaching GNSS and IMU sensors and geting their data

AMINTAHOUNI, 2022-10-30:
CARLA version: 0.9.13
Platform/OS: windows

I have generated a simple scenario in which the ego vehicle moves in a straight line and another vehicles moves in front of ego vehicle.
both vehicles are moving with a constant speed.

I want to get the speed and location of both vehicles using sensors. so,
1- how can I attach IMU and GNSS sensors to these vehicles?
2- how can I get data from these sensors.

I want to calculate time to collision (TTC).
the TTC = (X_L - X_F)/(V_F - V_L)

Can anyone help me?
Thanks

TzabarDolev, 2023-09-19:
I wrote a short article about IMUs and they're usage in CARLA in the Carla research blog - https://carlasimblog.wordpress.com/2023/09/16/understanding-imu-sensors-and-their-role-in-autonomous-driving-simulations-with-carla/.
Hope you find it helpful.

Blyron, 2026-11-12:
GNSS and IMU are documented in [ref_sensors.md](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/ref_sensors.md). For time-to-collision it is simpler to use `actor.get_location()` and `actor.get_velocity()` of both actors each tick. Closing.

closed 2026-11-12 by Blyron