carla-simulator/carla #5907: make PythonAPI error "Failed to find XercesC (missing: XercesC_VERSION)"

fawadshahid, 2022-11-03:
`BuildOSM2ODR.sh: Building OSM2ODR.
Cloning into '/home/gz0044/carla/Build/libosm2dr-source'...
remote: Enumerating objects: 1006678, done.
remote: Counting objects: 100% (13071/13071), done.
remote: Compressing objects: 100% (1772/1772), done.
remote: Total 1006678 (delta 12401), reused 11299 (delta 11299), pack-reused 993607
Receiving objects: 100% (1006678/1006678), 717.60 MiB | 2.29 MiB/s, done.
Resolving deltas: 100% (846531/846531), done.
Updating files: 100% (78068/78068), done.
Note: switching to 'ee0c2b9241fef5365a6bc044ac82e6580b8ce936'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at ee0c2b9241 Removed debug warnings
-- Setting build type to 'Release' as none was specified.
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Could not determine Eclipse version, assuming at least 3.6 (Helios). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_BINARY_DIR: /home/gz0044/carla/Build/libosm2dr-build
-- CMAKE_SOURCE_DIR: /home/gz0044/carla/Build/libosm2dr-source
-- 
-- Platform: 
--     Host: Linux-5.15.0-52-generic x86_64
--     Target: Linux-5.15.0-52-generic x86_64
--     CMake: 3.16.3
--     CMake generator: Ninja
--     CMake build tool: /usr/bin/ninja
--     Compiler: Clang 8.0.1
-- 
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindXercesC.cmake:99 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:71 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/gz0044/carla/Build/libosm2dr-build/CMakeFiles/CMakeOutput.log".
make: *** [Util/BuildTools/Linux.mk:152: osm2odr] Error 1
`
Get this during when building pythonAPI. I'm checked out to branch 0.9.12, I also came across https://github.com/carla-simulator/carla/issues/5305 issue which i fixedmy changing the Xerces version in the setup.sh

KaustubhKael, 2022-11-03:
Any progress on this issue? I am also stuck ate the same point.

HORIZON-ing, 2022-11-04:
I have solved the problem.  详细解决链接：https://zhuanlan.zhihu.com/p/579805147
CARLA安装时make PythonAPI遇到错误,，首先会报错网址404，因为xerces-c-3.2.3的版本已经更新为xerces-c-3.2.4，原始链接已经失效，新地址为:https://ftp.cixug.es/apache/xerces/c/3/sources/，可以手动下载后放在路径下手动安装，也可以打开Setup.sh，把版本3.2.3改为3.2.4。后面还是会报"Failed to find XercesC (missing: XercesC_VERSION)"的错误，需要把在路径下carla/Build下的xerces-c-3.2.4-install复制一份，重命名为xerces-c-3.2.3-install:

fawadshahid, 2022-11-04:
> I have solved the problem. 详细解决链接：https://zhuanlan.zhihu.com/p/579481065 CARLA安装时make [...]

Thank you!! this solved my problem @KaustubhKael

KaustubhKael, 2022-11-04:
@HORIZON-ing @fawadshahid , this worked, thank you!

MoolixLin, 2022-11-06:
@HORIZON-ing  I met some problem when I "make launch", the error "make: *** [CarlaUE4Editor] 错误 6" occured. Thr main problem is like this, have you met the problem like this?  I use UE4.26.2, carla0.9.13, Visual Studio 2019 14.29.30146 and Windows 10.0.18362.0 SDK.  I don't how to solve this problem exactly.
`[Upgrade]
[Upgrade] Using backward-compatible build settings. The latest version of UE4 sets the following values by default, which may require code changes:
[Upgrade]     bLegacyPublicIncludePaths = false                 => Omits subfolders from public include paths to reduce compiler command line length. (Previously: true).
[Upgrade]     ShadowVariableWarningLevel = WarningLevel.Error   => Treats shadowed variable warnings as errors. (Previously: WarningLevel.Warning).
[Upgrade]     PCHUsage = PCHUsageMode.UseExplicitOrSharedPCHs   => Set in build.cs files to enables IWYU-style PCH model. See https://docs.unrealengine.com/en-US/Programming/BuildTools/UnrealBuildTool/IWYU/index.html. (Previously: PCHUsageMode.UseSharedPCHs).
[Upgrade] Suppress this message by setting 'DefaultBuildSettings = BuildSettingsVersion.V2;' in CarlaUE4Editor.Target.cs, and explicitly overriding settings that differ from the new defaults.
[Upgrade]`

closed 2022-11-08 by fawadshahid

hugeshark7, 2023-05-06:
> 

thanks bro, 你解决了我的问题

SirMichcik, 2024-09-18:
> I have solved the problem. 详细解决链接：https://zhuanlan.zhihu.com/p/579481065
> CARLA安装时make [...]

Hey, could anyone give me a step by step instruction how to do this? I have downloaded new carla 0.9.13 version and uninstalled all xerces packages that i have installed previously