carla-simulator/carla #5914: Semantic Segmentation - Is it possible to subdivide the pedestrian class?

s-hd, 2022-11-04:
CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04.4 LTS

Hello,

I use the semantic segmentation sensor of CARLA, but noticed that riders of motorcycles, bikes and other 2-Wheelers are classified seperately from the 2-Wheelers as Pedestrians.

[image]

Is it possible to create 2 seperate classes of Pedestrians:
1) Pedestrians who are walking
2) Pedestrians who are driving 2-Wheelers

In issue #5003 the question was also asked, but not specifically answered. I managed to distinguish between 4-Wheelers and 2-Wheelers after instructions in #5003 and #175 . There, you can specifiy the path to the related folders in Tagger.h, but there is no distinction of pedestrians based on folder structure as far as I understand. Is there a possible solution for this distinction?

DnDare, 2022-11-28:
Greetings, and sry for delay. 
I think that best option is to duplicate the rider pedestrian, move the duplicate to a different folder of your choice, assign the new pedestrian to the vehicle, and create the new Semantic class for that folder. 

If you have doubts, ask again,
[image]

DnDare, 2022-11-28:
I've an update 
We've been working on a new semantic segmentation,

**Riders have their own semantic class,** 

We'll update this week, maximum next week,  

You just will have to update, is this good for you? else you can try the pipeline I proposed you.

Regards

s-hd, 2022-11-29:
Thank you very much for the answers and the suggested pipeline! Nice that there is an update coming. I will wait and try it out.

Regards

closed 2024-04-22 by DnDare