carla-simulator/carla #5920: Failed to start carla, directly end at startup without error messages

hrz2000, 2022-11-08:
Operating System: Ubuntu 16.04 LTS
Kernel: Linux 4.4.0-210-generic
Architecture: x86-64
carla version: CARLA_0.9.10.1

When Carla is started, it ends directly instead of keeping active (I didn't end it manually). The gpu does not display the corresponding activity. What is the possible reason? There is no error prompt. It is difficult for me to find the problem. 
When I try to run it on Docker, the same situation occurs. 
```bash
(pl) hrz@colab1:/data1/hrz/plant/carla$ SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh --world-port=2000 -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
(pl) hrz@colab1:/data1/hrz/plant/carla$ docker run -it --rm --net=host --gpus '"device=0"' carlasim/carla:0.9.10.1 ./CarlaUE4.sh --world-port=2000 -opengl
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
```
I know that the above outputs are normal, but the problem is that the program ends immediately after executing the instructions, and there is no error message. (On my other machine, the program can be executed normally and remains active. )
Has anyone ever encountered this problem？

hrz2000, 2022-11-16:
Thank you for your kind advice. I tried the 0.9.13 version of carla, but it also doesn't work also. 
[image]

Here are some descriptions of the machine I can't run
```
CPU: 
    CPU(s):                32
    Model name:            Intel(R) Xeon(R) CPU E5-2620 v4 @ 2.10GHz
    Stepping:              1
    CPU MHz:               1199.953
    CPU max MHz:           3000.0000
    CPU min MHz:           1200.0000

RAM: 
    MemTotal:       98873340 kB
    MemFree:         7035976 kB
    MemAvailable:   95738432 kB
    Buffers:          969412 kB
    Cached:         83881764 kB
    SwapCached:        16148 kB
    Active:         36203156 kB

GPU: 4 * [GeForce GTX 1080 Ti]
```

On the machine where I can successfully run carla, its configuration is as follows, 
```
CPU: Model name: Intel Core Processor (Haswell, no TSX, IBRS)
GPU: 4* GTX TITAN X
```
Thanks again for your kind answer~

Dragonking3x, 2022-11-16:
The hardware looks good.
It is hard to tell what the problem could be.

The only other thing that could be is the driver for Nvidia. (update driver to 510?)
It would be strange to have that problem with `-RenderOffScreen`, but you could try it.

Another Graphic related problem could be this
`https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#installation-guide`

I am running the docker with this line:
`docker run --privileged --gpus all --net=host -e DISPLAY=$DISPLAY -e SDL_VIDEODRIVER=x11 -v /tmp/.X11-unix:/tmp/.X11-unix:rw carlasim/carla:0.9.13 /bin/bash ./CarlaUE4.sh`

Maybe one of these fixes your problem, but I will keep thinking about a solution.

hrz2000, 2022-11-16:
Thank you again for your prompt reply！
machine which i can't run well:
```
Driver Version: 440.82       CUDA Version: 10.2 
```
machine which i can run well:
```
Driver Version: 418.87.00    CUDA Version: 10.1
```
So I didn't try to upgrade Nvidia drivers
> Another Graphic related problem could be this
https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/install-guide.html#installation-guide

I tried to install nvidia-container-toolkit as well as tried the startup command you gave, but carla is still not working properly.

Finally, thank you again for your warm reply~

EunhoCho, 2024-03-28:
Is this problem solved..? I have the same issue.

Blyron, 2026-11-12:
This report is about an old CARLA release on Unreal Engine 4, which is no longer maintained. Closing as obsolete. Please try CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`). If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron