carla-simulator/carla #5921: make PythonAPI, make launch errors in cpp compilation

blouxongit, 2022-11-08:
Hello everyone,

I am trying to build carla and am facing an issue when building it. I am following this guide : [https://carla.readthedocs.io/en/latest/build_linux/](linux_build)

CARLA version: just cloned https://github.com/carla-simulator/carla, should be CARLA 0.9.13
Platform/OS: Ubuntu 20.04
Unreal Engine 4.26
Python version : 3.8.10
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1) 

I have faced few issues that I already solved (which I will list in case it is relevant to my current problem).
I am now blocked at the 'make launch' and 'make PythonAPI' steps.

What I already did : 

-Modify the clang variables to not block in case of clang 8+ (see here) https://github.com/carla-simulator/carla/issues/5847#issuecomment-1286977595

-Modify url for wget appache/xerces (see here) https://github.com/carla-simulator/carla/issues/5305#issue-1187350244

When running both of the command lines mentionned above, I get the following results : 

`$ make pythonAPI`
[image]

`$ make launch`
[image]

Any ideas how to fix this ?
Thank you very much!

JosenJiang, 2022-11-09:
I get the same error. I am also looking for solutions.

JosenJiang, 2022-11-09:
This [issue](https://github.com/carla-simulator/carla/issues/2416) may help you

blouxongit, 2022-11-14:
> This [issue](https://github.com/carla-simulator/carla/issues/2416) may help you.
> 
> Modify test_streaming.cpp file (refer to the solution of ncointe): Line 58 and 91, replace "Server" by "carla::streaming::low_level::Server" Line 63 and 94, replace "Client" by "carla::streaming::low_level::Client"

Thank you it worked perfectly!