carla-simulator/carla #5923: #creating nuscenes format dataset#boundingbox

ZXJohn, 2022-11-09:
CARLA version: 9.13
Platform/OS:ubuntu20.04/windows
Problem I have experienced:
     1、When the hero vehicle is moving, I found the bounding box of other vehicle will deviates from the right position.
     2、The box will still export even it is blocked by buildings.
if anyone could give me some advance.

makaveli10, 2022-11-09:
Hi @ZXJohn 
You could remove the vehicles that are blocked by buildings 
- by filtering them with distance from the ego vehicle
- by using depth data and filtering out the vehicles with respect to that

As for the deviation, I think in my case that happens only when the ego vehicle is turning otherwise the bounding boxes are alright.

JosenJiang, 2022-12-26:
Hi @makaveli10, I have also met the same problems as @ZXJohn mentioned when I follow the [doc](https://carla.readthedocs.io/en/latest/tuto_G_bounding_boxes/#bounding-boxes). The bounding box of other vehicle will deviates from the right position. Would you mind giving some advice?

makaveli10, 2023-01-02:
@JosenJiang could you share some example?

JosenJiang, 2023-01-03:
> 

I found the cause of the problem. The image frame and the world frame is not the same. Thanks anyway.

makaveli10, 2023-01-04:
@JosenJiang awesome. I was under the impression that you are already in sync mode and checking if data from all sensors have the same frame as you get from ```world.tick()```
here is what i did:
```
frame = world.tick()
# assume data has all sensors raw data
assert all(x.frame == frame for x in data)
```

JosenJiang, 2023-01-04:
@makaveli10 You Inspire me! Thank you!

makaveli10, 2023-01-07:
@JosenJiang btw are you getting correct bboxes when you run your annotations through nuscenes-devkit?

Blyron, 2026-11-12:
As found in the comments, the drift comes from running asynchronously: use synchronous mode so the sensor data and actor transforms belong to the same frame. Occlusion needs the depth or instance segmentation image. See [Bounding boxes](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/tuto_bounding_boxes.md). Closing.

closed 2026-11-12 by Blyron