carla-simulator/carla #5924: Why is carla/Build/patchelf-install/ empty?

awesomekuei, 2022-11-09:
When I was running make PythonAPI, the error occurs like

[image]

Then what should I do , could someone tell me specify steps ? Thanks ！！

zeroyang0809, 2022-12-17:
i meet the same question ... how to solve
my ubuntu is 20.04

Blyron, 2026-11-12:
This is an error from the old Make/`Setup.sh` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake and uses the Unreal Engine toolchain. See [Build on Linux](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_linux_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron