carla-simulator/carla #5930: Does NVIDIA GeForce GTX 1050 adequate to install carla #5929

unrealtz, 2022-11-10:
Hi dear,
I have 2 questions :

    I have a NVIDIA GeForce GTX 1050 which has 4 GB GPU is it adequate to install carla?
    visual studio 2019 does not suport Windows 8.1 SDK so I install Windows 10 SDK, is it compatible to build Carla?

I would appreciate if anyone could help me

[Cook-2022.11.10-00.28.41.txt](https://github.com/carla-simulator/carla/files/9983916/Cook-2022.11.10-00.28.41.txt)
[Log.txt](https://github.com/carla-simulator/carla/files/9983917/Log.txt)

jokolaco, 2022-11-16:
Yes, it is possible to build carla with an GTX 1050 TI. But later if you ran the simulator you get very slow response. Even if you run the simulator in off-screen rendering mode the performance will be not optimal, but in the worse case you can use such a setup to render off-screen data into images.
The problem is more the "low" RAM than the speed of the GPU. So it is better to use at least a 6 GB RAM Card or better a 8 GB card, e.g. a GTX 1070 is fine.

unrealtz, 2022-11-17:
thank you jokolaco,
I have a new problem as follow: 

-[Package]: Building package...
Exception calling "Write" with "3" argument(s): "Exception of type 'System.OutOfMemoryException' was thrown."
At
C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules\Microsoft.PowerShell.Archive\Microsoft.PowerShell.Archive.psm1:820
char:29
+ ...                     $destStream.Write($buffer, 0, $numberOfBytesRead)
+                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], MethodInvocationException
    + FullyQualifiedErrorId : OutOfMemoryException

I would be appreciate if you help me to solve this.

jokolaco, 2022-11-17:
Sorry, i don´t use windows. 
But as mentioned in the feedback you need more RAM or diskspace.
Please read the docs!

closed 2024-05-17 by MarcelPiNacy-CVC