carla-simulator/carla #5933: Unrealistic black shadows in map from RoadRunner

jannine92, 2022-11-11:
CARLA version: 0.9.13 (build from sources)
Platform/OS: Windows 10

Hello,

I'm creating maps in RoadRunner (versoin: R2022a), export the .fbx and .xodr files and use them to import the map into CARLA.
Unfortunately, the elements throw shadows with unrealistic black areas that change when I'm moving to change the view.
When I edit the skylight and unclick the field "Cast shadows", the black areas are gone. That's ok as a workaround for my local installation but I also need to ingest the map in a Carla package where I cannot do that (because only the .fbx and .xodr files are used).

Has anyone had the same problem or knows how to fix it?

Here is a screenshot of the shadows with black areas:

[image]

This is how it looks like when I unclick "Cast shadows": 

[image]

KooMB2, 2022-11-21:
https://github.com/carla-simulator/carla/issues/3877

this might be help

DnDare, 2022-11-28:
Yes, as KooMB2 mentions, there's a workaround so you can deactivate them generally or specifically to geometries, we usually deactivate the ground/floor/roads because they generate "ugly" shadows.
It's an issue with the MeshDistanceFields, these are detected in areas  with fill light only (enviromental) and no direct light (sun).

closed 2023-02-01 by glopezdiest