carla-simulator/carla #5951: Generate Routes is false in Carla

aurorazhang13, 2022-11-18:
Hi,

I try to manually import a new map to Carla following the official docs, but I can't check Generate Routes after I checked Add Spawners. Does anyone know how to solve this problem?
[image]

CARLA version: 0.9.12 (build from source)
Platform/OS: Windows

fah870599552, 2023-02-21:
Have you solved this problem? I encoutered the same one

aurorazhang13, 2023-02-21:
> Have you solved this problem? I encoutered the same one

It seems that you don't have to check it.

fah870599552, 2023-02-21:
>兄弟，这个没有route会影响traffic manager的控制函数吗

Blyron, 2026-11-12:
This is about the Unreal Engine 4 `make import` pipeline, which no longer exists. In CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) maps are imported with `Util/Tools/Import.py` or shipped as content packs, see [Custom maps](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_maps.md) and [Content packs](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_packs.md). Closing as obsolete. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron