carla-simulator/carla #5976: How can I add a unreal plugin to carla?

ronyshaji, 2022-11-23:
Hello

How can I add a unreal plugin to the carla?
Any help would be appreciated.

DnDare, 2022-11-28:
Hi, you must have the UE4 project with carla's content, I guess you already have it. 
Best way would be to install from UE marketplace, some can be installed just finding the right version compatible with CARLA, usually 4.26 works. Using the add to project option.
Alternativelly, some plugins won't let you "add to project" option, just "create  project". This is a bit tricky, but you can create a new project and then migrate, or copy manually the content to CARLA's nstallation. Will not work 100% of projects, but it's worth trying . 

Some plugins are meant to be activated also from the Plugin manager:
https://docs.unrealengine.com/4.27/en-US/ProductionPipelines/Plugins/

If you have issues, tell me which plugin you want to add, so that I can review options.

ronyshaji, 2022-11-29:
@DnDare Thanks for the reply. Actually I created a custom blueprint (.uasset) in UE and need to import to CARLA so that I can control it using the carla python API. But copy the custom asset to the carla content folder doesnot work for me as it is not considered as an actor. Do you have any idea how can i import a custom asset to carla ? I thought of doing it through the plugin.

DnDare, 2022-11-29:
@ronyshaji  If I understood correctly, you tried copying manually from your custom source project through the file explorer (Windows or Linux?), and didn't work.  
Do you have acces to  your custom plugin from UE4 Carla's content browser? If you do I think you can manually (drag'n'drop) Copy, Move  or make an Advance Copy (this keeps links and redirectors).
I don't think you need to add new content through the plugin settings, neither from a custom marketplace asset. We never tried.

Have you tryed the Migrate option?
https://docs.unrealengine.com/4.27/en-US/Basics/AssetsAndPackages/Migrate/
Migrate is the best option, I think. You can add content from unrelated projects.

Regards

ronyshaji, 2022-11-29:
Yes i have a project in unreal engine and i just copied it into the carla but it doesnot work. It can be seen in carla but not able to control it as it is not considered as an actor.

DnDare, 2022-11-30:
> but not able to control it as it is not considered as an actor.
At this point I'll have to ask the core team,  
What kind of project are you working on? A new model or similar? Or more complex. A new actor, or something based on existing actor classes?

ronyshaji, 2022-12-01:
@DnDare My project is much simple. I created a blueprint/asset using procedural mesh in Unreal engine and have the .uasset file with me. I need to import that to carla and spawn the asset in the carla just like we spawn a vehicle or a traffic light. Regular copy and pasting in teh carla content folder doesnot work for me. This is the thing i am currently looking for...may be is there any other options ?

DnDare, 2022-12-01:
@ronyshaji  So you have a generated procedural mesh, created in a different independent project. 
If you just want to use the new mesh, you can try  select in Content Browser, RMB/AssetActions/Export
This way you export a fbx, then you should be able to reimport as a normal asset. 
Also you can try with a MergeActors before exporting, this way you get a brand new mesh to export.

If you want to import the BP procedural mesh generator you've created. I don't have enough info to help, but I don't find a specific reason, not all plpugins we tried to merge worked, but they usually do.

ronyshaji, 2022-12-01:
My blueprint is created in unreal engine and it has a .uasset format. When i asked unreal experts they said that it is not easy to convert the .uasset format to .fbx.

Also i tried to import the .uasset using the command 'Make Import' but it failed. Is there any way to add the asset using the c++ code. I have the c++ code of my asset ready.

Axel1092, 2022-12-02:
Hi @ronyshaji, it is hard to understand what you are trying to do. As @DnDare said, you can move assets from one project to another with the migrate functionality https://docs.unrealengine.com/4.27/en-US/Basics/AssetsAndPackages/Migrate/. But if this blueprint relies on custom C++ code (classes, functions, etc) then you need to bring the code to the project as well. If you already have a plugin created, then you can just copy it to the `Unreal/CarlaUE4/Plugins` folder. Otherwise, you will need to create the plugin and probably change the asset to match the newly created plugin. More info about the plugins again as @DnDare said https://docs.unrealengine.com/4.27/en-US/ProductionPipelines/Plugins/.

ronyshaji, 2022-12-05:
@Axel1092 @DnDare My project is simple. I made a procedural mesh blueprint in the unreal engine and have it in .uasset format. So I need to use that asset so that I can spawn in CARLA like a vehilce or traffic light using Python. 

So If i go through the plugin method, can I able tos pawn my custom blueprint asset using python ?

ronyshaji, 2022-12-15:
@DnDare Just a small clarification. If I add my asset using the plugins, do I able to spawn it via the carla Python API ?

ronyshaji, 2023-01-11:
@DnDare Can i use the added plugin in carla python API ? Any idea ?

DnDare, 2023-01-12:
If you want to add plugins, first you'll have to add them to UE. 
If you want to use plugins from the API, you should expose it's functionality to the API.

ronyshaji, 2023-01-12:
@DnDare I am sorry that the main title is confusing you. Let me explain my exact target.

**I made a custom asset (procedural mesh) in unreal engine and now I need to spawn it using carla Python API (world.try_spawn()) and as per documentation I can only spawn actors from carla library. So how can i add my new asset to the library or how can i spawn using carla python API ?**

Axel1092, 2023-01-12:
Hi @ronyshaji, if you just want a new mesh in your scene, you could convert the procedural mesh to static mesh and add it to the `DefinitionsMap` variable inside the `PropFactory` blueprint (in the `FillDefinitionsMap` function) so that you can spawn it with the API. However, if you plan to use more features of the procedural mesh component (like creating geometry in runtime) then you will have to create a new factory to spawn the actor (check sources inside `Unreal/CarlaUE4/Pluguins/Carla/Source/Carla/Actor/`) and the all functionality and API related to this new actor.

closed 2024-04-22 by DnDare