carla-simulator/carla #5986: Make PythonAPI failure - error MSB1009

junaidbinsarfraz, 2022-11-25:
CARLA version: 0.9.13
Platform/OS: Win 10 (latest)
Command: `make PythonAPI`

 Log/Error: 
```
HEAD is now at ee0c2b9241 Removed debug warnings
-- Selecting Windows SDK version 10.0.20348.0 to target Windows 10.0.19044.
-- CMAKE_BINARY_DIR: D:/carla/Build/osm2odr-visualstudio
-- CMAKE_SOURCE_DIR: D:/carla/Build/om2odr-source
--
-- Platform:
--     Host: Windows-10.0.19044 AMD64
--     Target: Windows-10.0.19044 AMD64
--     CMake: 3.25.0
--     CMake generator: Visual Studio 16 2019
--     CMake build tool: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
--     Compiler: MSVC 19.29.30147.0
--
CMake Error at C:/Program Files/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.25/Modules/FindXercesC.cmake:112 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:71 (find_package)

-- Configuring incomplete, errors occurred!
See also "D:/carla/Build/osm2odr-visualstudio/CMakeFiles/CMakeOutput.log".
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: Project file does not exist.
Switch: install.vcxproj
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "D:\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_wheel'

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\carla\PythonAPI\carla\dist"!
```

Sub issue of https://github.com/carla-simulator/carla/issues/21

hutomosaleh, 2022-11-26:
You need to run `pip install wheel`

junaidbinsarfraz, 2022-11-26:
> You need to run `pip install wheel`

Right. And what about this error

```
CMake Error at C:/Program Files/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.25/Modules/FindXercesC.cmake:112 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:71 (find_package)
```

hutomosaleh, 2022-11-26:
Seems related to this:  https://github.com/carla-simulator/carla/issues/5632
Are you at master branch? There's a lot of broken dependencies links there.
It might make more sense to switch to a stabler branch. Maybe try 0.9.13 `git checkout tags/0.9.13`

junaidbinsarfraz, 2022-11-26:
> Seems related to this: #5632 Are you at master branch? There's a lot of broken dependencies links there. It might make more sense to switch to a stabler branch. Maybe try 0.9.13 `git checkout tags/0.9.13`

Make sense, switched to `tags/0.9.13`. Now should I run `make clean` before `make PythonAPI`? Or how to clean previous build before building again?

hutomosaleh, 2022-11-26:
yes `make clean`. You can also delete the build folder. But make sure to change the .bat files for zlib and xerces, since the download paths are outdated.

junaidbinsarfraz, 2022-12-04:
> Seems related to this: #5632 Are you at master branch? There's a lot of broken dependencies links there. It might make more sense to switch to a stabler branch. Maybe try 0.9.13 `git checkout tags/0.9.13`

After `git checkout tags/0.9.13` should I have to run `git pull origin tags/0.9.13` before running `make clean` or `make PythonAPI`?

danijel-fmci, 2022-12-13:
Hi, 
I did same as [junaidbinsarfraz](https://github.com/junaidbinsarfraz) and I am still getting error after executing `make PyhtonAPI:`

`BuildLibCarla.sh: Success!
BuildOSM2ODR.sh: Building OSM2ODR.
Cloning into '/home/fmci_carla/carla/Build/libosm2dr-source'...
remote: Enumerating objects: 1006678, done.
remote: Counting objects: 100% (13071/13071), done.
remote: Compressing objects: 100% (1772/1772), done.
remote: Total 1006678 (delta 12401), reused 11299 (delta 11299), pack-reused 993607
Receiving objects: 100% (1006678/1006678), 717.52 MiB | 15.43 MiB/s, done.
Resolving deltas: 100% (846534/846534), done.
Updating files: 100% (78068/78068), done.
Note: switching to 'ee0c2b9241fef5365a6bc044ac82e6580b8ce936'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at ee0c2b9241 Removed debug warnings
-- Setting build type to 'Release' as none was specified.
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Could not determine Eclipse version, assuming at least 3.6 (Helios). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_BINARY_DIR: /home/fmci_carla/carla/Build/libosm2dr-build
-- CMAKE_SOURCE_DIR: /home/fmci_carla/carla/Build/libosm2dr-source
-- 
-- Platform: 
--     Host: Linux-5.4.0-135-generic x86_64
--     Target: Linux-5.4.0-135-generic x86_64
--     CMake: 3.16.3
--     CMake generator: Ninja
--     CMake build tool: /usr/bin/ninja
--     Compiler: Clang 8.0.1
-- 
-- Found **XercesC: /usr/lib/x86_64-linux-gnu/libxerces-c.so (found version "3.2.2")** 
-- Found Proj: /home/fmci_carla/carla/Build/proj-install/lib/libproj.a
-- Found X11: /usr/include   
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Enabled features: Linux-5.4.0-135-generic x86_64 Clang 8.0.1 Release Proj
-- Configuring done
-- Generating done
-- Build files have been written to: /home/fmci_carla/carla/Build/libosm2dr-build
/home/fmci_carla/carla/Util/BuildTools/BuildOSM2ODR.sh: line 103: -DXercesC_INCLUDE_DIR=/home/fmci_carla/carla/Build/xerces-c-3.2.4-install/include: No such file or directory
make: *** [Util/BuildTools/Linux.mk:152: osm2odr] Error 127`

Platform/OS: Ubuntu 20.04

junaidbinsarfraz, 2022-12-13:
This is for windows, but I guess you can do similar in Ubuntu as well. Try [this solution](https://github.com/carla-simulator/carla/issues/3320#issuecomment-735540901).

danijel-fmci, 2022-12-13:
> This is for windows, but I guess you can do similar in Ubuntu as well. Try [this solution](https://github.com/carla-simulator/carla/issues/3320#issuecomment-735540901).

Thank you junaidbinsarfraz 

I solved the issue by steps provided here [https://github.com/carla-simulator/carla/issues/5305#issuecomment-1317102835](url)

This way is changed XERCESC_REPO link for XERCESC_VERSION=3.2.3

Nitro60zeus, 2023-05-22:
@gfresdoc did you solve it?

gfresdoc, 2023-08-30:
> @gfresdoc did you solve it?

@Nitro60zeus I switched to version .14 and I manged to make it working. Same steps as the guide but it worked.

If you can, I recommend building under Linux. Under Linux building is much less problematic.

If you cannot try using .14

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron