carla-simulator/carla #5989: make launch,error, unresolved external symbol

olyii, 2022-11-25:


olyii, 2022-11-25:
[1/4] UE4Editor-MathWorksUAV.dll
    正在创建库 D:\UnrealEngine\carla\Unreal\CarlaUE4\Plugins\Marketplace\MathWorks\MathWorksSimulation\Intermediate\Build\Win64\UE4Editor\Development\MathWorksUAV\UE4Editor-MathWorksUAV.suppressed.lib 和对象 D:\UnrealEngine\carla\Unreal\CarlaUE4\Plugins\Marketplace\MathWorks\MathWorksSimulation\Intermediate\Build\Win64\UE4Editor\Development\MathWorksUAV\UE4Editor-MathWorksUAV.suppressed.exp
  LINK : error LNK2001: 无法解析的外部符号 IMPLEMENT_MODULE_MathWorksUAV
  Sim3dFixedWing.gen.cpp.obj : error LNK2019: 无法解析的外部符号 "public: __cdecl ASim3dFixedWing::ASim3dFixedWing(void)" (??0ASim3dFixedWing@@QEAA@XZ)，函数 "void __cdecl InternalConstructor<class ASim3dFixedWing>(class FObjectInitializer const &)" (??$InternalConstructor@VASim3dFixedWing@@@@YAXAEBVFObjectInitializer@@@Z) 中引用了该符号
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl ASim3dFixedWing::Sim3dInit(void)" (?Sim3dInit@ASim3dFixedWing@@UEAAXXZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl ASim3dFixedWing::Sim3dSetup(void)" (?Sim3dSetup@ASim3dFixedWing@@UEAAXXZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl ASim3dFixedWing::Sim3dStep(float)" (?Sim3dStep@ASim3dFixedWing@@UEAAXM@Z)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl ASim3dFixedWing::Sim3dRelease(void)" (?Sim3dRelease@ASim3dFixedWing@@UEAAXXZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl ASim3dFixedWing::Transform(void)" (?Transform@ASim3dFixedWing@@UEAAXXZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dActor::AllocateTransformArrays(void)" (?AllocateTransformArrays@ASim3dActor@@MEAAXXZ)
  Sim3dQuadRotor.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dActor::AllocateTransformArrays(void)" (?AllocateTransformArrays@ASim3dActor@@MEAAXXZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dActor::BeginPlay(void)" (?BeginPlay@ASim3dActor@@MEAAXXZ)
  Sim3dQuadRotor.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dActor::BeginPlay(void)" (?BeginPlay@ASim3dActor@@MEAAXXZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dVeh::ClaimMainCamera(void)" (?ClaimMainCamera@ASim3dVeh@@MEAAXXZ)
  Sim3dQuadRotor.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dVeh::ClaimMainCamera(void)" (?ClaimMainCamera@ASim3dVeh@@MEAAXXZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual class ASim3dActor * __cdecl ASim3dVeh::CreateInstance(struct Simulation3DCreateActorStruct const *,class TSubclassOf<class AActor>,class ASim3dActor *,class UWorld *)const " (?CreateInstance@ASim3dVeh@@MEBAPEAVASim3dActor@@PEBUSimulation3DCreateActorStruct@@V?$TSubclassOf@VAActor@@@@PEAV2@PEAVUWorld@@@Z)
  Sim3dQuadRotor.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual class ASim3dActor * __cdecl ASim3dVeh::CreateInstance(struct Simulation3DCreateActorStruct const *,class TSubclassOf<class AActor>,class ASim3dActor *,class UWorld *)const " (?CreateInstance@ASim3dVeh@@MEBAPEAVASim3dActor@@PEBUSimulation3DCreateActorStruct@@V?$TSubclassOf@VAActor@@@@PEAV2@PEAVUWorld@@@Z)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dVeh::CreateVehicleConfigsTopicReader(void)" (?CreateVehicleConfigsTopicReader@ASim3dVeh@@MEAAXXZ)
  Sim3dQuadRotor.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dVeh::CreateVehicleConfigsTopicReader(void)" (?CreateVehicleConfigsTopicReader@ASim3dVeh@@MEAAXXZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dVeh::DeleteVehicleConfigsTopicReader(void)" (?DeleteVehicleConfigsTopicReader@ASim3dVeh@@MEAAXXZ)
  Sim3dQuadRotor.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "protected: virtual void __cdecl ASim3dVeh::DeleteVehicleConfigsTopicReader(void)" (?DeleteVehicleConfigsTopicReader@ASim3dVeh@@MEAAXXZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl ASim3dActor::EndPlay(enum EEndPlayReason::Type)" (?EndPlay@ASim3dActor@@UEAAXW4Type@EEndPlayReason@@@Z)
  Sim3dQuadRotor.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl ASim3dActor::EndPlay(enum EEndPlayReason::Type)" (?EndPlay@ASim3dActor@@UEAAXW4Type@EEndPlayReason@@@Z)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual struct FVector __cdecl ASim3dActor::GetSim3dAngularVelocity(void)const " (?GetSim3dAngularVelocity@ASim3dActor@@UEBA?AUFVector@@XZ)
  Sim3dQuadRotor.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual struct FVector __cdecl ASim3dActor::GetSim3dAngularVelocity(void)const " (?GetSim3dAngularVelocity@ASim3dActor@@UEBA?AUFVector@@XZ)
  Sim3dFixedWing.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual struct FRotator __cdecl ASim3dActor::GetSim3dRotation(void)const " (?GetSim3dRotation@ASim3dActor@@UEBA?AUFRotator@@XZ)
  Sim3dQuadRotor.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual struct FRotator __cdecl ASim3dActor::GetSim3dRotation(void)const " (?GetSim3dRotation@ASim3dActor@@UEBA?AUFRotator@@XZ)
[... cut ...]
  Sim3dMWAirliner.gen.cpp.obj : error LNK2019: 无法解析的外部符号 "public: virtual void __cdecl ASim3dAerospace::TransformReturn(int,int)" (?TransformReturn@ASim3dAerospace@@UEAAXHH@Z)，函数 "public: virtual void * __cdecl ASim3dAerospace::`vector deleting destructor'(unsigned int)" (??_EASim3dAerospace@@UEAAPEAXI@Z) 中引用了该符号
  Sim3dSkyHogg.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl ASim3dAerospace::TransformReturn(int,int)" (?TransformReturn@ASim3dAerospace@@UEAAXHH@Z)
  Sim3dAerospace.gen.cpp.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl ASim3dAerospace::TransformReturn(int,int)" (?TransformReturn@ASim3dAerospace@@UEAAXHH@Z

olyii, 2022-11-25:
[image]

Blyron, 2026-11-12:
This is a link error in the third-party MathWorks UAV plugin on the Unreal Engine 4 build, not in CARLA. Closing, as the UE4 build is no longer maintained.

closed 2026-11-12 by Blyron