carla-simulator/carla #5990: How to show the view in the rearview mirror

ergolab006, 2022-11-28:
[image]

We want to show the view in the rear-mirror ， we take a lot of time to deal with problem，but I don't find the solution of my problem . Could anyone help me , thanks !!

rengimozokutgen, 2023-02-20:
I'm trying to do a similar thing (trying to add side mirrors as well as rear-mirror), but couldn't find the solution yet. I hope someone could help us.

YiguangM, 2023-04-29:
Why is no body willing to help?

greendonut5, 2023-06-29:
Hi I also want to implement this, as well as side mirrors. Hopefully somebody helps out

inrkumar22, 2023-09-06:
I am also looking for the solution. I have tried the below solutions but not succussed with the expected result.
I made the side mirrors of one of the car model (Audi A2) as reflective surface using UnrealEditor 4.27.2. In unreal editor it shows the real reflective surface. But the same kind of reflective object is not captured on the side mirror. It captures blurred images when running the same car model with manual_script.py in Pygame Windows and running Carla simulator in UnrealEngine.

Can anyone suggest any input here. Any kind of help is really appreciated.

Tingsouu, 2023-09-07:
Add new sensor,  surface, and Pygame  built-in class CameraManager(object):

Blyron, 2024-05-10:
Adding this kind of stuff is quite complicated, you will have to do editor changes . We have not implemented this yet. 
https://www.youtube.com/watch?v=kglwTYXTUV8&ab_channel=IvicaMilari%C4%87
This is not in our plans in short terms
We are gonna close this issue