carla-simulator/carla #6004: Xerces error

shekharsuman3, 2022-12-02:
Platform:
--     Host: Windows-10.0.19042 AMD64
--     Target: Windows-10.0.19042 AMD64
--     CMake: 3.21.3
--     CMake generator: Visual Studio 16 2019
--     CMake build tool: D:/programs/Microsoft visual studio/visual studio/MSBuild/Current/Bin/MSBuild.exe
--     Compiler: MSVC 19.29.30146.0
--
-- Found Proj: D:/carla/Build/proj-install/lib/proj.lib
-- Found ZLIB: D:/carla/PythonAPI/carla/dependencies/lib/zlib.lib (found version "1.2.13")
-- Enabled features: Windows-10.0.19042 AMD64 MSVC 19.29.30146.0 Release Proj
-- Configuring done
-- Generating done
-- Build files have been written to: D:/carla/Build/osm2odr-visualstudio
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  osm2odr.vcxproj -> D:\carla\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  -- Install configuration: "Release"
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "D:\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\u27e45\AppData\Local\Programs\Python\Python37\include -IC:\Users\u27e45\AppData\Local\Programs\Python\Python37\include "-ID:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-ID:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" "-ID:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-ID:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:dependencies/lib /LIBPATH:C:\Users\u27e45\AppData\Local\Programs\Python\Python37\libs /LIBPATH:C:\Users\u27e45\AppData\Local\Programs\Python\Python37\PCbuild\amd64 "/LIBPATH:D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\ATLMFC\lib\x64" "/LIBPATH:D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x64" "/LIBPATH:D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\ATLMFC\lib\x86" "/LIBPATH:D:\programs\Microsoft visual studio\visual studio\VC\Tools\MSVC\14.29.30133\lib\x86" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x86" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x86" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\um\x86" /EXPORT:PyInit_libcarla build\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj /OUT:build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.lib shlwapi.lib Advapi32.lib ole32.lib shell32.lib D:\carla\PythonAPI\carla\dependencies/lib\carla_client.lib D:\carla\PythonAPI\carla\dependencies/lib\Detour.lib D:\carla\PythonAPI\carla\dependencies/lib\DetourCrowd.lib D:\carla\PythonAPI\carla\dependencies/lib\libboost_filesystem-vc142-mt-x64-1_72.lib D:\carla\PythonAPI\carla\dependencies/lib\libboost_python37-vc142-mt-x64-1_72.lib D:\carla\PythonAPI\carla\dependencies/lib\osm2odr.lib D:\carla\PythonAPI\carla\dependencies/lib\proj.lib D:\carla\PythonAPI\carla\dependencies/lib\Recast.lib D:\carla\PythonAPI\carla\dependencies/lib\rpc.lib D:\carla\PythonAPI\carla\dependencies/lib\sqlite3.lib D:\carla\PythonAPI\carla\dependencies/lib\xerces-c_3.lib D:\carla\PythonAPI\carla\dependencies/lib\zlib.lib
   Creating library build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.lib and object build\temp.win-amd64-3.7\Release\source/libcarla\libcarla.cp37-win_amd64.exp
libcarla.obj : error LNK2001: unresolved external symbol png_destroy_write_struct
libcarla.obj : error LNK2001: unresolved external symbol png_set_bKGD
libcarla.obj : error LNK2001: unresolved external symbol png_set_cHRM_fixed
libcarla.obj : error LNK2001: unresolved external symbol png_set_pHYs
libcarla.obj : error LNK2001: unresolved external symbol png_set_swap_alpha
libcarla.obj : error LNK2001: unresolved external symbol png_set_text
libcarla.obj : error LNK2001: unresolved external symbol png_set_swap
libcarla.obj : error LNK2001: unresolved external symbol png_set_invert_mono
libcarla.obj : error LNK2001: unresolved external symbol png_set_sRGB
libcarla.obj : error LNK2001: unresolved external symbol png_set_compression_mem_level
libcarla.obj : error LNK2001: unresolved external symbol png_get_io_ptr
libcarla.obj : error LNK2001: unresolved external symbol png_create_write_struct
libcarla.obj : error LNK2001: unresolved external symbol png_set_gAMA_fixed
libcarla.obj : error LNK2001: unresolved external symbol png_write_info
libcarla.obj : error LNK2001: unresolved external symbol png_set_compression_buffer_size
libcarla.obj : error LNK2001: unresolved external symbol png_set_tIME
libcarla.obj : error LNK2001: unresolved external symbol png_write_row
libcarla.obj : error LNK2001: unresolved external symbol png_set_pCAL
libcarla.obj : error LNK2001: unresolved external symbol png_set_PLTE
libcarla.obj : error LNK2001: unresolved external symbol png_set_sBIT
libcarla.obj : error LNK2001: unresolved external symbol png_set_oFFs
libcarla.obj : error LNK2001: unresolved external symbol png_set_filter
libcarla.obj : error LNK2001: unresolved external symbol png_set_iCCP
libcarla.obj : error LNK2001: unresolved external symbol png_set_hIST
libcarla.obj : error LNK2001: unresolved external symbol png_set_compression_level
libcarla.obj : error LNK2001: unresolved external symbol png_destroy_read_struct
libcarla.obj : error LNK2001: unresolved external symbol png_set_IHDR
libcarla.obj : error LNK2001: unresolved external symbol png_set_strip_alpha
libcarla.obj : error LNK2001: unresolved external symbol png_set_compression_strategy
libcarla.obj : error LNK2001: unresolved external symbol png_set_write_fn
libcarla.obj : error LNK2001: unresolved external symbol png_create_info_struct
[... 45 lines cut ...]
proj.lib(networkfilemanager.obj) : error LNK2001: unresolved external symbol sqlite3_reset
proj.lib(factory.obj) : error LNK2001: unresolved external symbol sqlite3_reset
proj.lib(factory.obj) : error LNK2001: unresolved external symbol sqlite3_libversion
proj.lib(factory.obj) : error LNK2001: unresolved external symbol sqlite3_libversion_number
[... cut ...]
proj.lib(sqlite3_utils.obj) : error LNK2001: unresolved external symbol sqlite3_vfs_unregister
proj.lib(proj_json_streaming_writer.obj) : error LNK2001: unresolved external symbol sqlite3_vsnprintf
build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd : fatal error LNK1120: 79 unresolved externals
error: command 'D:\\programs\\Microsoft visual studio\\visual studio\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\carla\PythonAPI\carla\dist"!

LucasFKobernic, 2023-01-18:
Hi @shekharsuman3 did u find a solution?

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron